MEGA65/mega65-core #409: make simulate error

udaydeepu, 2021-06-19:
error: unit "arch8x6w16n4i10k4fci4fco8pfi8pfo8iopb2_wrapper_8bit" not found in library "work"  
I tried to change in the Makefile and still getting the error.

gurcei, 2021-06-21:
Not sure on your specific error, but maybe you can elaborate on your situation and we might get some ideas.

Are you using the latest 'development' branch of mega65-core? Which OS are you on?

I recently braved getting ghdl to build in windows, so have some experience with getting "make simulate" to work, so maybe we can get an idea by comparing our efforts/directions a bit.

gurcei, 2021-06-21:
Oh, I also noticed that ticket #398 has a similar experience.

gurcei, 2021-07-01:
Ok, as we haven't heard back anything on this ticket, and there are users that have successfully used "make simulate" on 'development' branch recently, I'll close this ticket.

closed 2021-07-01 by gurcei