MEGA65/mega65-core #411: synth_1 failed run for mega65r3.bit make

clar0867, 2021-06-20:
Unable to make mega65r3.bit as synth_1 run fails preventing impl_1 to launch.
[image]

gurcei, 2021-06-20:
Which branch are you building from? If not using it already, I'd suggest working from "development" branch, as that is the most active branch at present and is building successfully

clar0867, 2021-06-21:
Build was from the development branch.

gurcei, 2021-06-21:
Ah ok, so just for point of comparison, our Jenkins build server successfully built the latest commit to 'development' branch (commit#73a2cab) from 3 days ago.

So if you're building from this exact commit too and it is failing, it'll require figuring out what is different on your system compared to what the jenkings build agent has.

If you've made changes to the vhdl since then, you might need to scrutinise those changes.

Also might be worth reaching out to falk on the discord channel to learn more details on what pre-reqs are on those build agents and how they differ to you.

E.g. Could it be a difference in versions of vivado causing this? (Though I've switched between a few different versions of vivado in my time and had success, but still, can't rule it out).

Also, jist to share my own recent experiences, I built from latest development branch on my local ubuntu vm last night, for `make bin/mega65r3.bit` and `make bin/nexys4ddr-widget.bit`, and both built fine.

gardners, 2021-06-21:
This can also be caused by having insufficient RAM. Vivado needs _lots_ of
RAM, probably at least 4GB, and more likely 8GB to successfully synthesise
the MEGA65.

Paul.

clar0867, 2021-06-22:
Appears that the solution to running make bin/mega65r3.bit using Vivado 2020.2 is to run an excess of 8GB of RAM and multiple processor cores to enable this. I feel as though a minimum inclusion of hardware requirements would be beneficial as this has solely come down to a lack of information on this behalf. 
If aware I could have resolved this issue weeks ago throwing near full capacity of 32GB RAM and the i7-10875H into the VM for this to conduct this almost instantly.

gurcei, 2021-06-22:
Glad you got it sorted, and can sympathise with the pains you went to get things working.

Also good to see that not long after reaching out via the ticket, you got some answers that helped make up for the shortcomings of present documentation.

So moving forward, I agree with you that it'd be good to document minimum hardware requirements for building bitstreams somewhere:
- speed of cpu
- number or cores
- how much hard disk space is needed for the vivado installation
- how much memory

As for where a good home for these details might be, I was thinking perhaps in the repo's build.md file?

https://github.com/MEGA65/mega65-core/blob/master/docs/build.md

Perhaps @gardners might have another home for it in mind?

@clar0867, if you're keen, you're welcome to add such details and open a PR for us to merge in.

Also might be good for it to point to a walkthrough of how to install vivado, as I recall someone recently tripping up on install steps.

Someone has already prepared some docs for vivado steps, I will try to integrate it into the mega65 user guide soon.

https://github.com/MEGA65/mega65-user-guide

gurcei, 2021-07-01:
Ok, on this matter, I've played my part and merged in Amok's documentation on installing-vivado into the latest mega65-book.pdf file.

I'll leave this ticket open a little longer, to see if you're open to adding those minimum-hardware-details into the 'build.md' document.

gurcei, 2021-07-13:
I've found a location inside the mega65-book.pdf file where some mention of the required specs was made for synthesising bitstreams, in "CH19: Flashing the FPGAs and CPLDs in the MEGA65":

> Be prepared to install many gigabytes of software on a Linux or Windows PC, before you
will be able to write programs for the FPGAs and CPLDs in the MEGA65. Also, ”compiling”
complex designs can take up to several hours, depending on the speed and memory capacity
of your computer. We recommend a computer with at least 8GB RAM (preferably
16GB) if you want to write programs for FPGAs and CPLDs.

I will update this paragraph to reflect your recent experiences, and perhaps provide a concise list of suggested pre-reqs for synthesis:

- **speed of cpu**: As fast as you can get your hands on!
- **number or cores**: The free license for Vivado can make use of a maximum of 8 parallel cores (so ideally, 8 or a few more)
- **how much hard disk space is needed for the vivado installation**: Several GB, depending on how much you install (bear in mind the full install file is 50GB)
- **how much memory**: Latest bitstreams need a minimum about 12GB of RAM to build (ideally, have more, to play it safe)

gurcei, 2021-07-16:
I've found a location inside the mega65-book.pdf file where some mention of the required specs was made for synthesising bitstreams, in "CH19: Flashing the FPGAs and CPLDs in the MEGA65":

> Be prepared to install many gigabytes of software on a Linux or Windows PC, before you
will be able to write programs for the FPGAs and CPLDs in the MEGA65. Also, ”compiling”
complex designs can take up to several hours, depending on the speed and memory capacity
of your computer. We recommend a computer with at least 8GB RAM (preferably
16GB) if you want to write programs for FPGAs and CPLDs.

I will update this paragraph to reflect your recent experiences, and perhaps provide a concise list of suggested pre-reqs for synthesis:

- **speed of cpu**: As fast as you can get your hands on!
- **number or cores**: The free license for Vivado can make use of a maximum of 8 parallel cores (so ideally, 8 or a few more)
- **how much hard disk space is needed for the vivado installation**: Several GB, depending on how much you install (bear in mind the full install file is 50GB)
- **how much memory**: Latest bitstreams need a minimum about 12GB of RAM to build (ideally, have more, to play it safe)

Ok, I feel like `mega65-book.pdf` has been updated sufficiently well now to address your feedback, feel we can close this ticket off.

closed 2021-07-16 by gurcei