MEGA65/mega65-core #412: "mfm_bits_to_gaps" in shell script

amar0039, 2021-06-20:
How we can solve the "mfm_bits_to_gaps"  error for creating shell script.
We can solve it manually in Makefile.  
Add:- $(VHDLSRCDIR)/mfm_bits_to_gaps.vhdl \

gurcei, 2021-06-20:
I saw HarlodHolt commited a solution for this recently that got merged into master branch.

https://github.com/MEGA65/mega65-core/commit/9080fe3420b3b93b34aebb37283a95e67717702b

As I'm currently on development branch, this fix hasn't arrived on that branch yet, but I've added the same fix to mt current PR which will hopefully make it to development branch in the coming days

gurcei, 2021-06-21:
Feel like this situation has been explained, so will close the ticket

closed 2021-06-21 by gurcei