MEGA65/mega65-core #414: General issues

Dipesh-Regmi, 2021-06-20:
I am interested in this program however one of the difficulties for me I don't know which package is needed to install. I have installed all the packages that are mention but still get errors. So, I need to search in google and the required package. and run the command. This happens every time I run a command like Make simulate. Then I again need to search for that package and install it. Another issue is I don't have enough storage to install vivado. Vivado is a huge file. So I can't be able to do the whole task as required. Is there any other small file (like vivado) that help me to continue my work.

rdpeake, 2021-06-20:
Regarding Vivado at least - it is a required component as it is the compiler for the VHDL codebase - there is no way around installing that: see #408

gurcei, 2021-06-21:
With rdpeake explaining the situation with Vivado, that leaves open the topic you raised of possible undocumented dependencies.

Perhaps you can document here any packages you needed that weren't documented.

Also, keep note that the most active branch right now is the 'development' branch (which is hopefully the branch you're working from), so if such pre-reqs aren't documented there, then we should add them to the docs.

If you're keen, you're welcome to add these pre-reqs to the docs and open a PR.

gurcei, 2021-07-01:
Would have been good to capture any missing pre-reqs in the docs, but as there's no response on this, think it's time to close the ticket.

closed 2021-07-01 by gurcei