MEGA65/mega65-core #415: Pixel poo appearing in bitplane mode(s) at bottom left and top left corners

deftmega, 2021-06-26:
One pixel top left and a small row bottom left can be spotted, please check with attached disk image

[image]

[image]

[MEGADEMO.zip](https://github.com/MEGA65/mega65-core/files/6720107/MEGADEMO.zip)

deftmega, 2021-07-12:
Pixel poo is related to CRT emulation! No poo if CRT emulation is disabled. please fix crt emulation.

lydon42, 2021-12-19:
Shorter test:
10 SCREEN 320,200,1
20 LINE 319,0,319,0:REM DIRT LOWER RIGHT
30 LINE 0,10,0,10:REM DIRT UPPER LEFT
40 GETKEY A$
50 IF A$="X" THEN GOTO 80
60 POKE $D054,PEEK($D054) XOR 32:REM TOGGLE CRT
70 GOTO 40
80 POKE $D054,PEEK($D054) AND 223:REM DISABLE CRT
90 SCREEN CLEAR

closed 2022-01-09 by lydon42