MEGA65/mega65-core #418: Interesting things going on with lines and possibly circles in BASIC

deftmega, 2021-06-26:
please see attached images for reference, disk images also attached
[image]
[image]
[image]
[MEGADEMO.zip](https://github.com/MEGA65/mega65-core/files/6720119/MEGADEMO.zip)

Edilbert, 2021-06-27:
I think, this effect is generated by selecting the "raster effect" on HDMI display.

deftmega, 2021-06-28:
negative, I always watch on VGA while capturing (the above) on HDMI

deftmega, 2021-06-28:
correction: you are of course right Edilbert for certain effects happening. But I was referring to double lines on left side (bottom screenshot), extra pixels on left side (top screenshot), and doubled lines on top (middle and bottom screenshot).

Edilbert, 2021-06-28:
I see. That is probably related to the pixel poo issue.
It seems, that the scanline from memory, that is transferred to display, is not properly aligned.
There seems to be an offset of 1 pixel. This would explain effects on the left and the right border.

deftmega, 2021-07-12:
Pixels appearing on left border is related to CRT emulation! also see issue #415. Please fix crt emulation.

!!!Only duplicated top vertical line remains as an issue if CRT emulation is turned off.!!!

frehwagen, 2021-12-19:
Top issue is the same as in  #432

closed 2021-12-20 by deftmega