MEGA65/mega65-core #422: Saving CONFIGURE settings under nexys

kernal-clink, 2021-07-05:
When using latest bitstream (nexys4ddr-widget-20210628.22-develo-400b006.bit) settings are unable to be saved

gurcei, 2021-07-11:
Oh, I'm wondering if this relates to something that user Amok experienced early on too.

I was reading his past threads on Discord recently and saw that he switched:

- https://discord.com/channels/719326990221574164/719326990221574168/842697905290215435

FROM: loading the bitstream from the sd-card (the default path nexys users are suggested to use in our docs)

TO: loading the bitstream from the QSPI flash chip (which is a less trodden path and not as well described in our docs, but improving)

I read that the possible reasons that made him switch were:

1) Problems rebooting after onboarding finished or freeze-menu used to swap PAL/NTSC
2) Such settings not being saved.

So it's possible that settings not being saved from the configuration utility might also be affected by this, and so you might want to try switching to QSPI bitstream-loading mode yourself.

I'm underway with merging Amok's documentation on his nexys4 experiences of topics like this (transitioning over to using QSPI). Some of these efforts have made their way into the latest mega65-book.pdf

Temporary location of latest documentation is here:
* http://gurce.net/mega65_files/docs/

In "CH10: Using Nexys4 boards as a MEGA65", there's now a section called "Loading the bitstream from QSPI".

I've still got two more docs from Amok to merge in that probably have more details too, so will put *all* of his docs on your radar too, just in-case.

https://drive.google.com/drive/folders/1gCMJYNMlzHUNEyNNFBCIPwh8zOnVaL__?usp=sharing

gurcei, 2021-07-11:
I also had a read of the discord thread that spawned this ticket:

- https://discord.com/channels/719326990221574164/783728598313992262/861702464189235230

The thread is a bit intermixed with another thread, but Paul jumped in at this point providing some reasoning:

- https://discord.com/channels/719326990221574164/783728598313992262/861761354692296724

> gardners — 06/07/2021
> Kernal adtbm - the problem here is almost certainly in the configure programme incorrectly guessing which SD card bus to write to.  You are welcome to dig into the configure programme and see if this is indeed the issue. Its in asm.
> 
> There is code that selects which SD card bus.  It might be that we need to check the target ID register, and use the opposite bus by default on Nexys to all other platforms.

gardners, 2021-07-12:
Configuration goes to SD card, and it should not matter at all whether the bitstream was loaded from QSPI, USB or carrier pigeon. Only flashing QSPI is restricted if you didn't load the bitstream from QSPI (for complicated Xilinxy reasons that are possibly possible to overcome, but I haven't figured out how to).

headkaze, 2021-07-17:
Be aware that the default setting for QSPI ConfigRate in [nexys4.xdc](https://github.com/MEGA65/mega65-core/blob/master/src/vhdl/nexys4.xdc#L267) and [nexys4ddr.xdc](https://github.com/MEGA65/mega65-core/blob/master/src/vhdl/nexys4ddr.xdc#L241) is 66MHz:

`
set_property BITSTREAM.CONFIG.CONFIGRATE 66 [current_design]
`

But it may be too fast so you may need a lower setting like 26MHz:

`
set_property BITSTREAM.CONFIG.CONFIGRATE 26 [current_design]
`

gardners, 2021-07-17:
Correct -- but now that the configure settings are not stored in the QSPI, but rather on the SD card. But you are quite right, that on some boards at least, 26MHz may be required, if the system doesn't boot reliably from QSPI.

gurcei, 2021-07-24:
Just refreshing my memory on what I observed from testing @gardners' attempted fix, mentioned in this discord thread:
- https://discord.com/channels/719326990221574164/783728598313992262/866247419490336769

Seems like it didn't work as hoped, that settings weren't getting saved successfully to sector 1, and visibility of the sd-card during startup seemed a little hit-and-miss (with occasional hickup reports of `NO MEGA65.ROM FOUND` on some startups).

I'll dig into it some more tonight...

gurcei, 2021-07-27:
Ok, a fix for this has been merged into 'development' branch

PR #428 (2021-07-27): #422: Saving CONFIGURE settings under nexys
Resolves #422.
closed 2021-07-27 by gurcei