MEGA65/mega65-core #437: Saving to disc images no longer works with latest bitstream

steph72, 2021-08-28:
With bitstream development@292846b it is no longer possible to save files to disc images mounted from the internal SD card (using closed ROM V920218)

Steps to reproduce:
1. Mount a .d81 disc image from the freezer (or have it auto mounted at startup)
2. Create an arbitrary BASIC one liner
3. save it with `DSAVE "FOO"`
4. type `DIR`
5. "foo" wasn't saved and doesn't show up in the directory

[image]

steph72, 2021-08-28:
P.S.: Issue is not related to BASIC65. Using CBDOS through the C64 mode kernal has the same results.

gardners, 2021-08-28:
Roger that. Will investigate tonight. Most likely still something funny with implementation of floppy writing.

lgblgblgb, 2021-08-28:
I had similar problems with Xemu in the old days when I have a ( .... err ... more ) bugs in the fdc emulation. It's really interesting though that there is no error message or anything that it could not be written, this was my experience as well.

gardners, 2021-08-28:
Confirmed that I can save to a D81 on commit 80a1b3c, and also on ccd8468.

@steph72 can you please double-check that there is no HICKUP.M65 on the SD card? As there was a bug in the HICKUP code that would cause this problem.

There were also some recent commits where it was broken, but it should not be 100% fine.

steph72, 2021-08-29:
Problem disappears with a freshly formatted SD card.

closed 2021-08-29 by steph72

lgblgblgb, 2021-08-29:
Btw it seems **Gurce found the possible reason that deleted file is picked up** still, that is on filename comparison uses `DEX` followed by `BNE` and not `BPL` so the first (zeroth ...) char of the name is never checked. Among other problems (like picking up also `AICKUP.M65` for example instead of `HICKUP.M65`, it also causes to happy find a deleted file as exactly the first character of the filename is overwritten by `$E5` in that case, which is not checked then by that bug.