MEGA65/mega65-core #441: VIC displays wrong border colour in 2-bitplane mode and even colour indexes.

Edilbert, 2021-09-10:
10 FOR BC=0 TO 7:BORDER BC
20 SCREEN 320,200,2
30 PRINT BC,PEEK($D020)
40 GETKEY A$
50 SCREEN CLOSE
60 NEXT BC

This test program opens in a loop a graphics screen for the border colours 0 to 7.
The border colours 2, 4 and 6 are displayed as 1, 2 and 3, while the printout shows the correct numbers.
The other indexes are correct.
0 -> 0
1 -> 1
2 -> 1
3 -> 3
4 -> 2
5 -> 5
6 -> 3
7 -> 7

So if bit 0 of the border colour is 0, the displayed colour is the register colour shifted right.

lydon42, 2021-12-20:
Can't reproduce this with current development@0f9af7e nor master@bbd0391 on nexys4ddr.
Border colours are correct, return values of peek match BC.
Is this still an issue? Please clarify platform, core and rom used.

lydon42, 2021-12-30:
Possibly fixed by 32 color patch. Problem no longer exists. Bit Shifter confirms.

closed 2021-12-30 by lydon42