MEGA65/mega65-core #449: Enhance SID, please?

M3wP, 2021-09-25:
Please make some enhancements to the SIDs including extra oscillators and better noise, as discussed.

This link was requested to be included:  `https://justanotherelectronicsblog.com/?p=402`

M3wP, 2021-11-30:
> I need to add support for the Nexys 4 to NetSID<https://github.com/headkaze/NetSID> as I think it will be useful for improving the SID support in MEGA65.
>
> NetSID allows you to play SIDs (with 2SID and 3SID support) via serial UART using the (ACID 64 player)[https://www.acid64.com/].

Perhaps convert it to LAN based protocol?  You have the source?

Daniel.

headkaze, 2021-12-01:
> Perhaps convert it to LAN based protocol? You have the source?

Building the entire MEGA65 bitstream just to work on the SID implementation would be quite time consuming so I propose using NetSID for faster development. You can make modifications to the vhdl, build a bitstream directly to the Nexys 4, then test any song using the ACID 64 player over UART.

The history of the SID implementation used in MEGA65 is quite interesting as referenced in the [README.md](https://github.com/headkaze/NetSID/blob/main/README.md). Essentially we don't know the original author of the vhdl but we do know that [alvieboy](https://github.com/alvieboy) added filter support.

I was under the impression this implementation was quite complete? What features are missing or are you proposing enhancements not contained in the original SID chip?

gardners, 2022-01-26:
Relevant reference material for accurate SID emulation (in no particular order):

https://www.kvraudio.com/forum/viewtopic.php?t=526480  -- includes SPICE model of the filter. Can we use that to implement a VHDL one?
https://bel.fi/alankila/c64-sw/index-cpp.html
http://forum.6502.org/viewtopic.php?f=8&t=4150

There are some tools for converting SPICE to Verilog, for example, which could be a nice way to tackle this. E.g.:

https://www.tekmos.com/about/blog/spice-to-verilog

deftmega, 2022-01-29:
Let's try Sorgelig's new Mister SID for a change:

deftmega, 2022-01-29:
https://github.com/MiSTer-devel/C64_MiSTer/tree/master/rtl/sid

deftmega, 2022-01-29:
oh, one more comment: we need someone (NOT Paul) to dig into porting reSID to VHDL in full. It has not been done because the other projects don't have big enough FPGAs - we do. This can be the perfect SID without the limitations of a too small FPGA, like FPGASID. If anyone dares to take this job, please contact us / me for support <3

deftmega, 2022-01-30:
More valuable resources:
http://www.sidmusic.org/sid/sidtech.html
https://bel.fi/alankila/c64-sw/index-cpp.html

deftmega, 2022-01-31:
from discord: 

 Test bit (from the inSIDeous document):
On every synthesizer, the oscillators are usually running in the background continuously. If there is silence, the oscillators will still be playing, but you won't be able to hear them because the amplitude envelope has reduced their volume to zero. When you press a key to play a note, the envelope allows the sound to pass through, but the oscillator itself does not restart its wave from the beginning. The test bit allows you to do just that. When it is set, the oscillator will have its wave restarted from the beginning (at a phase of zero). This is an advanced feature that is mostly useful when using multiple channels to create a single monophonic sound. If you set one channel to use the test bit and another to not, the two waves will have a random phase difference on every new note. You can also set the test bit at any point in the step table, allowing you to make it sound like a single waveform is playing a new note.
Try using 2 channels, both set to use the sawtooth wave and make sure the TEST bit is disabled for both. Play some notes to hear how it sounds. The sound should be consistent on each new note, but the phase between the 2 channels will be different. You can see this on the output scope at the bottom-right of inSIDious. Now enable the TEST bit on one channel. Play some notes again and you'll be able to hear (and see in the scope) that the phase is different between the two on every note. This is because one channel has its phase reset on every note while the other doesn't. Now enable the TEST bit on both channels. Because both channels have their phase reset on every note, the two waveforms are identical and so sound like a single channel at twice the volume.

`:
  reg8 test_next = control & 0x08;

  // Test bit set.
  // The accumulator and the shift register are both cleared.
  // NB! The shift register is not really cleared immediately. It seems like
  // the individual bits in the shift register start to fade down towards
  // zero when test is set. All bits reach zero within approximately
  // $2000 - $4000 cycles.
  // This is not modeled. There should fortunately be little audible output
  // from this peculiar behavior.
  if (test_next) {
    accumulator = 0;
    shift_register = 0;
  }
  // Test bit cleared.
  // The accumulator starts counting, and the shift register is reset to
  // the value 0x7ffff8.
  // NB! The shift register will not actually be set to this exact value if the
  // shift register bits have not had time to fade to zero.
  // This is not modeled.
  else if (test) {
    shift_register = 0x7ffff8;
  }

  test = test_next;`

M3wP: I suggest making test a counter and while not 0, fade the registers and decrement, setting it to some value on detecting the bit being set and unset (to stop the counter)

deftmega, 2022-01-31:
TEST (Bit 3)--The test bit, when set to a one, resets and locks oxcillator 41 at zero until the test bit is cleared. The noise waveform output of oscillator 41 is also reset and the pulse waveform output is held at a DC level. Normally this bit is used for testing purposes, however, it can be used to synchronize oscillator 41 to external events, allowing the generation of highly complex waveforms under real-time software control.

deftmega, 2022-01-31:
I found a complete reference on digital sampling and the C64, including: general theory, SID hardware description, and methods of playback, here:
http://sid.kubarth.com/articles/the_c64_digi.txt

closed 2024-09-10 by lydon42