MEGA65/mega65-core #452: Make a 15khz video output for older monitors

intric8, 2021-09-30:
This could allow one to use an old C= CRT monitor, for example, while also using an HDMI modern monitor. (which would be totally rad)

gardners, 2023-12-25:
Speaking with @ki-bo , we explored some of the solution space, and came to the following idea:

1. This should work with a special DB9 to DB15 adapter cable on the MEGA65's VGA port.
2. It should use the same cable as the M2M framework for MEGA65.
3. The cable should be used to provide the means of detection for the MEGA65 core (and M2M cores when they are updated to support this), so that it is impossible to accidentally provide 31KHz video to an old CRT monitor.
4. The VGA SDA/SCL pins are a good candidate for this: Bridge them using a 4.7K resistor, and the MEGA65 can check for this physical connection. If present, then the MEGA65 can assume it has a 15KHz monitor attached, and switch to 15KHz output.
5. The 15KHz video generation can be provided by r3_expansion.vhdl, that already generates 15KHz timed video for composite / S-video output. 
6. The R,G, and B channels are already the right voltage on VGA, so we just need to provide a "composite sync" signal, which we can already generate in r3_expansion.vhdl.
7. r3_expansion.vhdl should just gain VGA SCL/SDA pins, and a pass-through for the VGA from the core to the VGA connector, and do all the magic switching internally.
8. When working, coordinate with M2M framework team for them to support the same mode.

Ideally we should collate links to all the relevant resources in this issue. Some already:
https://www.retrogamingcables.co.uk/composite-video-vs-composite-sync
https://bboah-hardware.de/cgi-bin/showhardware.cgi?HARDID=849

gardners, 2023-12-25:
Probably me to do the VHDL parts. But I don't have a 1084 or similar monitor here, so will require assistance with testing.  It would be good for me to have a test cable with the resistor fitted, but with a momentary-break switch in-line for testing the auto-detection via watching sync on oscilloscope. i.e., so that I can watch the sync frequency change.

markkrueg, 2024-03-12:
It would be nice if there were a software "force mode" option as well. With warnings if necessary (although 15KHz mode shouldn't damage a modern VGA monitor). 

Currently I connect my MEGA65 to my US NTSC 1084S-P using a DB15 to 6 pin DIN connector. The US 1084 does not have SCART or a DB9, so the normal way to connect an Amiga or similar would be over the analog 6 pin DIN connector.

Anyway, I use this cable to connect my MEGA65 to my 1084, but currently it only operates when using the C64 core. This core has a nice menu where you can choose 15KHz mode and what type of sync you prefer. I'm hoping for a similar capability in the MEGA65 core without having to build another cable. 

https://www.ebay.com/itm/124200576180?var=425049611391
[image]

markkrueg, 2024-03-12:
Here is a comparison of PAL and NTSC models of 1084:
[image]

ZorroDJ, 2024-06-21:
Pease 115Khz RGB, not CVBS.

gardners, 2025-09-06:
I'm currently looking at this, using a VGA to SCART passive cable I have here.

Also fixing NTSC colour space issues for composite output on the expansion board.

gardners, 2025-09-06:
Confirmed that this gets 15KHz SCART output working for me from the VGA connector with a standard MiSTer cable.

matjam, 2026-01-19:
OK so I have the resistor thing working, you need a 470Ω resistor. Instructions are in #934. Would love some testing feedback.

[image]

commit 5fb15cdb (2025-09-06): more synth error fixes #452