MEGA65/mega65-core #453: Sprite alpha blending does not work?

M3wP, 2021-10-03:
[TestAlpha.zip](https://github.com/MEGA65/mega65-core/files/7274643/TestAlpha.zip)

I cannot get the attached test program to function as expected.  The sprites should blend with the characters, not the screen background colour?

In addition, if the sprites are instead set to appear behind the characters, they are not draw behind the 16bit characters where there is a $00 in the pixel data.  This problem may need a separate issue raised.

M3wP, 2021-10-04:
Oh I should say that if the effect _is_ meant to be the blending of the sprites with the background, then this only works for colour $00.  The rest give a flashing sprite effect.

M3wP, 2021-12-19:
You will need to manually poke the relevant addresses with values to see the problem/s if using the attached program.

lydon42, 2021-12-22:
I talked to Paul about this. It should work like this:
- one sprite pixel is determined (highest sprite, non transparent)
- this one pixel is blended with the pixel of the background using the alpha setting
- result is rendered
This does obviously not happen currently.
But important part is: there will be no alpha blending of multiple sprites stacked.
We did not talk about how this should interact with RBB alpha and sprite foreground/background settings.

M3wP, 2021-12-22:
On the foreground/background thing, I think having the transparency on should automatically make the sprites appear in the background.  This is just my two cents but I have considered the problem and it seems to me it would be the simplest logic?

I'm concerned about how it is working with non background character data at present since it is different between 16 and 256 colour tiles/cells/characters.  Automatically appearing in the background may fix this?