MEGA65/mega65-core #454: Fetch current CPU bank

M3wP, 2021-10-04:
When MAPping the CPU, it becomes important for library code to know where a seemingly near address is actually located in the flat memory space (for example, when declaring DMA job lists and trying to call them).

Being able to fetch the actual current bank will alleviate many issues when attempting to build code libraries.

M3wP, 2021-10-04:
[test454.zip](https://github.com/MEGA65/mega65-core/files/7281653/test454.zip)

I have created a test which tests the following conditions:

1. DMA started from $D706 works correctly in RAM when _not_ MAPped
2. DMA started from $D706 works correctly in RAM when MAPped

It adheres to the automated test harness requirements and reports test activity.

It succeeds on the R2 but apparently fails on the Nexys.  I am yet to confirm.

lydon42, 2021-12-21:
This is in hdfloppy and hwsc branches.
I just tested it on hwsc and get a green border, no 1 or 2 displayed. But I don't see any unit test output in my m65 tool.

M3wP, 2021-12-21:
I had to port the unit test output code from C to ASM.  You may be unwilling to debug it.  

If there is a green border, then it is working.

I do apologise about the unit test output unit not working.  I ran out of time and didn't know how to test it properly.

lydon42, 2021-12-22:
Is your issue resolved by the provided solution?

M3wP, 2021-12-22:
Umm...  Well...  The change does allow the DMA to be used more sensibly when utilising the MAP CPU feature.  I was hoping for a more generic solution (reading from the DMA perhaps) but the issue can be closed.

closed 2022-01-09 by lydon42