MEGA65/mega65-core #472: Floppy RDATA line can contain glitches

gardners, 2021-11-07:
Root cause unknown, as it suddenly started happening on my R3, prior to which reading was rock-solid.
Suspect VCC to floppy is too low at 4.7V.
The symptom is very short gaps, typically ~1/3 of the minimum interval length breaking up the true interval (as compared from multiple reads of the same track.
Thus the problem is believed to be in the read electronics. Symptoms seem consistent with a brief flux inversion being reported that is not really there.
Thus these short gaps should be treated as glitches and merged with the gaps on either side.

gardners, 2021-12-30:
Confirmed that it is _not_ on the read side, as disks written by other systems read without glitches.
Must therefore be on the write side.
R3A board has 5.07V instead of 4.65V on 5V line, so that is not the problem.
Suspect the combinatorial logic that selects the F_WDATA, F_WGATE etc is the most likely source of glitches.
Trying to move that to clocked to resolve this.

lydon42, 2022-01-09:
Floppy writes where tested thoroughly the last week, so this is fixed.

closed 2022-01-09 by lydon42