MEGA65/mega65-core #478: Cannot call hypervisor traps that use the X register as a parameter

jimnicholls, 2021-12-13:
Attempting to select the SD card partition ($00:$06) always fails regardless of the value in the X register. This includes if the X register is the default partition number.

The demonstration routine below uses service $00:$02 to get the default partition number and the uses service $00:06 to select the default partition. This should succeed regardless of the particular environment.  However the routine always returns with the A register having the value of $80. This value means "An attempt was made to select or operate on a disk or partition that does not exist".

```
        ; Get the default partition number
        lda     #$02
        sta     $d640
        nop
        ; Select the default partition
        tax
        lda     #$06
        sta     $d640
        nop
        ; If not successful, get and return the error code
        bcs     @10
        lda     #$38
        sta     $d640
        nop
        .byte   $2c     ; skip the next 2-byte instruction
        ; If successful, return 0
@10:    lda     #0
        rts
```

## Code path

(hwsc branch at commit a390dfe1e7d3ec043c8706737389b8cebe2ddcea)

The path through the hyppo code in executing service $00:06 is as follows.

```
trap_entry_points:
        jmp dos_and_process_trap
```
```
dos_and_process_trap:
        sei
        cld
        and #$7E
        tax
        jmp (dos_and_process_trap_table,x)
```
```
dos_and_process_trap_table:
        !16 trap_dos_getversion
        !16 trap_dos_getdefaultdrive
        !16 trap_dos_getcurrentdrive
        !16 trap_dos_selectdrive
        ; [Jim - the rest of dos_and_process_trap_table omitted for brevity]
```
```
trap_dos_selectdrive:
        jsr dos_set_current_disk
```
```
        dos_errorcode_no_such_disk = $80  ; [Jim - copied here from constants.asm for reference]
dos_set_current_disk:
        lda #dos_errorcode_no_such_disk
        sta dos_error_code
        cpx dos_disk_count
        bcc +
        jmp partitionerror
        ; [Jim - the rest of dos_set_current_disk does not execute]
```
```
partitionerror:
        ; [Jim - debugging code omitted for brevity]
        clc
        rts
```

## Observations

(Observed running xemu hyppo branch at commit lgblgblgb/xemu@0a9cda642d7ee1ed6dea1af079ce9367e5b00259 using m65dbg v1.00).

In `dos_set_current_disk` at `cpx dos_disk_count`, I observed the X register to be always 6, regardless what the X register was when the trap was triggered. I observed `dos_disk_count` to be 3, which matches the number of partitions in my SD card image file.

## Root cause analysis

`dos_and_process_trap` transfers the A register (function selector 6) into the X register (was the partition number) in order execute `jmp (dos_and_process_trap_table,x)`.

The X register is not saved or restored.

## Impact

This issue is not restricted to service $00:$06. Any of the $00:xx services that use the X register as a parameter are affected.

Similar code can be found in `memory_trap` and `syspart_trap`. Any of the $01:xx and $02:xx services are also similarly affected.

SydneyUni-Jim, 2021-12-14:
From @lgblgblgb: The X register is saved by the CPU in the hypervisor's view of $d461.

jimnicholls, 2021-12-14:
From @lgblgblgb: The X register is saved by the CPU in the hypervisor's view of $d461.

SydneyUni-Jim, 2021-12-19:
This issue can be closed. It was resolved by merging in #479, GitHub didn't think the issue to the PR.

jimnicholls, 2021-12-19:
This issue can be closed. It was resolved by merging in #479, GitHub didn't link the issue to the PR.

PR #479 (2021-12-19): Restore X before using it in dos_selectdrive and dos_cdroot.
Fixes #478
commit cc526e88 (2021-12-19): Restore X before using it in dos_selectdrive and dos_cdroot.

Fixes #478.
closed 2021-12-19 by jimnicholls