MEGA65/mega65-core #499: Faulty D076 register behavior?

hernandp, 2021-12-22:
Using the latest core and BASIC ROM yields the following behavior:  the default sprite mode seems to be full vertical resolution (V400) but poking POKE $D076,0 to disable it does not seem to work.

This can be tested with:

* Boot up the machine in 65 mode.
* Type:

10 FORI=1536TO1536+63:POKEI,255:NEXT
20 POKE$D015,255:POKE$D000,100:POKE$D001,100

run the program you will see the sprite in non-VICII mode (full vertical resolution).

POKE $0D76 to enable or disable the setting.  It wont change.

lydon42, 2021-12-22:
Tested on nexysA7 with latest master, development and hwsc plus ROM 920271 and 920273:
$D076 is always 0 on startup, and sprite is big, so V400 disabled.
Best report issues with specific core build (either jenkins number or git hash) and ROM version.

hernandp, 2021-12-22:
Cold resetting made it to work, but the start of my adventure into this is that in my Sprite Editor I get now V400 enabled at default, so I get bad cursor and preview sprites ;  but I couldn't reset it poking to D076, so I'm researching what  the Freezer environment changed in the last commits.  I know VICIV registers are setup differently now.

hernandp, 2021-12-22:
I reproduced it again by:

* Cold reboot
* Execute BASIC program, should work.
* Enter Freezer.
* Do not exit with F3/F5, Warm reset with Nexys board.
* Execute BASIC program, sprite is of incorrect height. D076 seems to be unoperational.

ROM 920273.
Freezer commit 16518d87ceea2d5bbf267495bf34aa87fb7831b6 
MEGA65 hypervisor 00.16, Development 20210810.19,3e0bacf

hernandp, 2021-12-22:
Updated to Development 20211229.22, 0F9AF7E.

Same behavior.

lydon42, 2021-12-23:
Just tested with 0f9af7e and 920273 on nexysA7, following your instructions: behaves as it should. Tested both PAL and NTSC modes.
Could you generate a md5sum of your FREEZER.M65? Perhaps the problem lies there. Just tested again with a freshly build FREEZER from hwsc head, but it also does not show the problem. My original FREEZER must be about two month old from the SD-Card Essentials file.
`b09b6066e0b05635834e6310307ce5eb  FREEZER.M65` (installed from SD Card Essentials)

Correction about the instructions:
- Power on
- _flash bitstream with m65 -b_
- run test program from disc
- enter freezer
- preset reset button on nexys
- run test program from disc

hernandp, 2021-12-23:
My FREEZER.M65 is built on the latest commit.
$ md5sum FREEZER.M65
5a03e8d7a998d209880c58fc2131a3c7 *FREEZER.M65

lydon42, 2021-12-23:
ok, now tried with the head of mega65-freezemenu: still no bug! Damn, the second to last commit message even mentions D076... I will look at the changes if something pops up... and the question would be: is this really a mega65-core issue? or is it a mega65-freezemenu issue?

hernandp, 2021-12-23:
@lydon42  Thanks for following up.
Made a quick video of it, sorry for the video shake (phone-in-hand) :)

https://youtu.be/lFlbbGjhWlQ

lydon42, 2021-12-23:
Already the first POKE$D076,255 looks wrong to me. The sprite should not be at the top position right below the border. rather it's distance in V200 mode it has to the top should be halved. So that's already suspicious.

This is how it looks for me: https://youtu.be/4cfVrJC87vo
You also see that my Freezer is way to low on the screen... only happens with NTSC.

hernandp, 2021-12-23:
Thanks. I can confirm that works in NTSC, but not in PAL.

gardners, 2022-01-02:
@hernandp please tell me how to reproduce the problem in the sprite editor, and I'll debug it directly from there.

hernandp, 2022-01-03:
@gardners Checkout  f655bcb437d4190597436cc28270d5a9776fbd8e from mega65-freezer (I added poking $D076 to zero in the startup code) and build FREEZER.M65 and SPRITED.M65 from there.
Upload to SD Card.

I'm using: 
ROM 920273.
Core: Development 20211219.22, 0F9AF7E.

Bootup machine. Start freezer -> and Sprite Editor. You'll see distorted cursor and preview sprite.

hernandp, 2022-01-03:
Addendum:  Attach the D076 test program. 
[D076.zip](https://github.com/MEGA65/mega65-core/files/7801746/D076.zip)

lydon42, 2022-01-20:
I could reproduce this using master@1586ad4 and 920287 on r3a.
Poking $d076 to 255 is not a prerequisite to trigger the bug, just FREEZE and then press reset button.
After this sprites are in the described state.

So this is nothing that is triggered by D076. It is some state then freezer sets the VIC to which is not cleared by a reset. Only way to get it back to normal seems to be power cycle.

The problem is only triggered by pressing the hardware reset button while in freezer. Pressing F3 or F5 in the freezer to reset does not trigger the problem.

lydon42, 2022-01-20:
I already compared VIC registers before and after the reset, but I could not find anything obvious. And setting VIC regs to values from before the reset did also have no effect on the wrongly displayed sprites.

Symptoms are:
- sprites are not in the position they should be (they are moved left and down compared to before)
- sprites are in V400 mode regardless of D076 contents

lydon42, 2022-01-20:
Steps to reproduce:
- have a prg that displays a sprite (the one above or my SPV4002.PRG)
- start in mega65 mode
- execute prg to see sprites
- freeze with restore
- press warm reset button of hardware
- execute prg
- bug!

Also:
- start sprite editor from freezer, sprites a are wrong

Reproduced with master@1586ad4 and 920287 on r3a.

[spv4002.zip](https://github.com/MEGA65/mega65-core/files/7908723/spv4002.zip)

lydon42, 2022-02-05:
No core bug, this is a freezer bug -> https://github.com/MEGA65/mega65-freezemenu/issues/33

lydon42, 2022-02-05:
This needs still an update of the mega65-freezemenu submodule, so reopening this as a reminder.

lydon42, 2022-02-06:
Pushed update to new freezemenu version.

closed 2022-02-06 by lydon42