MEGA65/mega65-core #505: HW Math Regs - division is off by a bit

lydon42, 2021-12-30:
Reported by Bit Shifter on discord
https://discord.com/channels/719326990221574164/782759365064917065/924985030227402795

In short: division is sometimes off by 1 bit. It seems that the lsb is always set
`          -- No idea why we need to add one, but we do to stop things like 4/2`
`          -- giving a result of 1.999999999`
`          q <= nn(63 downto 0) + 1;`

But as nn is 95 bits long, this decision could be rather made depending on the next bit?
Paul also mentioned that this could also related to the multiplication algorithm.

lydon42, 2022-01-03:
Getting unsure if the result really is wrong for 0x01 / 0x32...

Result:
>>> 0x028f5c28/2**32
0.009999999776482582
>>> 0x028f5c29/2**32
0.010000000009313226

Distance from 0.01:
>>> 0.01-0x028f5c28/2**32
2.2351741811588166e-10
>>> 0.01-0x028f5c29/2**32
-9.313225537987968e-12

As we can see the result the implemented algorithm gives us is actually nearer to 0.01 than the result we think we are looking for.

gardners, 2022-01-03:
Quite so. It might just not be "bug compatible" with the C64 BASIC floating point implementation.

lydon42, 2022-01-03:
Considering this conversion algorithm for the fractional part of a number:
```
def frac2bin(frac):
  bin = ''
  while len(bin)<32:
    frac = frac * 2 
    if frac > 1.0:
      frac -= 1.0
      bin += '1' 
    else:
      bin += '0' 
  return bin, frac
```
This will give us the results we see from old basic. But we will also see that the remainder fractional part will be >= 0.5 (next bit would be 1) for the results that are 1 bit "too much" in the HW Mult and < 0.5 (next bit 0) for the results that are equal.

So the HW Mult does give the more accurate result.

lydon42, 2022-01-03:
Considering this conversion algorithm for the fractional part of a number:
```
def frac2bin(frac):
  bin = ''
  while len(bin)<32:
    frac = frac * 2
    if frac > 1.0:
      frac -= 1.0
      bin += '1'
    else:
      bin += '0'
  return hex(int(bin,2)), frac
```
This will give us the results we see from old basic. But we will also see that the remainder fractional part will be >= 0.5   (next bit would be 1) for the results that are 1 bit "too much" in the HW Mult and < 0.5 (next bit 0) for the results that    are equal.

```
>>> 110/10000
0.011
>>> frac2bin(0.011)
('0x2d0e560', 0.25599999725818634)
-- HW Mult $0000006e / $00002710 = $0000000002d0e560
>>> 1/100
0.01
>>> frac2bin(0.01)
('0x28f5c28', 0.9600000008940697)
-- HW Mult $00000001 / $00000064 = $00000000028f5c29
```

So the HW Mult does give the more accurate result.

lydon42, 2022-01-09:
This is solved by the fix. Test will be added in the linked issue.

commit 89d255bc (2022-01-05): Merge pull request #517 from MEGA65/505-div-onebit

Resolves #505 - division is off by a bit
closed 2022-01-09 by lydon42