MEGA65/mega65-core #506: Math Acceleration Registers tests

lydon42, 2021-12-30:
Placeholder issue for the math acceleration register unit test.
- #248 
- #354 
- #505

Todo:
- [ ] check correctness of results
- [ ] check busy flag function

gardners, 2022-01-01:
Please provide exact hex inputs to put in to the divider that result in incorrect results.
Also are there values that result in incorrect multiplication? The reason I ask is that division is performed by finding the closest reciprocal by multiplying the reciprocal by the numerator to try to obtain the denominator (if I remember the details correctly), thus incorrect multiplication could be the root-cause.

lydon42, 2022-01-07:
vunit test is *very* good for debugging. Will add a "proper" in-system unit test anyways.