MEGA65/mega65-core #514: Update Makefile and Documentation for release

lydon42, 2022-01-03:
The Makefile needs a bigger cleanup and also some of the documentation files need an update.
This bundles a bunch of issues.

Tasks:
- [ ] CX65 variables should only refer to the binary, move options into CX65OPTS variables
- [ ] pass CX65 variables to submodules (freezemenu, fdisk), so that they don't need to build cc65, too
- [ ] cleanup requirements in docs/build.md
  - [ ] remove or update versions (python 2.X not cool!)
  - [ ] ImageMagick
  - [ ] libtinfo
  - [ ] exomizer

lydon42, 2022-01-09:
Force submodule Makefile execution: 6b57441e1fa8e56ac7506bc55969b9744848ff90

dansanderson, 2024-07-01:
I rewrote docs/build.md in August 2023, so those checkboxes can be ticked. https://github.com/MEGA65/mega65-core/blob/development/docs/build.md

lydon42, 2024-07-02:
And I did check all the dependencies for 0.96, so those can also be checked.