MEGA65/mega65-core #523: Ethernet RX buffer handling problems

gardners, 2022-01-16:
The ethernet controller seems to have problems with the RX buffer handling.  Initial investigation shows duplicate frames being received, suggesting that the wrong buffer is being presented to the CPU.  Similarly lost frames and seemingly out-of-order received frames suggests that this might be an issue.

The duplication of received frames was confirmed by enabling logging of received and transmitted ethernet frames in mega65-weeip, and then using a utility to convert those logs to normal pcap-compatible capture logs that could be analysed using tcpdump or wireshark.

gardners, 2022-01-16:
Maybe make an ethernet loop-back enabling program, to see if we can reproduce the problem.

gardners, 2022-06-30:
Confirmed that the ethernet buffer handling in VHDL gets confused if all buffers are filled (which of course can happen quite easily for client software written in C).

ki-bo, 2023-05-25:
This is probably solved by this commit: https://github.com/MEGA65/mega65-core/pull/653/commits/70522820ad65217f5d681a60566b4591e2982429

closed 2024-09-10 by ki-bo