MEGA65/mega65-core #526: N and V flags with BIT #imm

jimnicholls, 2022-01-19:
In the VHDL, does the immediate addressing mode of `BIT` (opcode 89) alter the N and V flags and should it?
Ref: http://forum.6502.org/viewtopic.php?f=2&t=2241&p=27243#p27239

Xemu and hyppotest both implement this quirk of `BIT`.

gardners, 2022-01-19:
Yes, N, V and Z are currently set for all version of the BIT instruction, including immediate mode. This is the simplest logic.  We can change it if its wrong, however.

jimnicholls, 2022-01-19:
There's a some ambiguity around if the real 4510 did this or not, and if 45GS02 needs to follow.

lgblgblgb, 2022-01-19:
> There's some ambiguity around if the 65CE02/4510 did this or not, and if 45GS02 needs to follow. I'm going to write a simple test program and see if someone with a 4510 can run it.

Even if 4510 does not follow, I think it's more important to be compatible with 65C02 at this point. Eg, I would guess about zero programs ever written for C65 which really depends on this. However 65C02 and compatible CPUs are used in many things, many compiler/assembler can be targeted that, and some 65C02 based stuff can be expected to be ported to MEGA65, with the promise that 65CE02 is in fact an enhanced 65C02 (thus the letter "E", I think). So just by chance, I think it's much more important to follow the standard. After all, some (early? most? all? no idea ...) 4510s known to exist in development C65s **could not even** execute well the Z-indexed addressing mode because some hardware bug in 4510 implementations. Does it mean that we must follow 4510 and forget Z indexed addressing mode then? ;)

_Surely it was my personal opinion only_. Of course if there is a much more major difference 4510 did differently (compared to lake 65C02) then it's already a bit different story.

lydon42, 2022-01-20:
Bo Zimmermann did run a test on the real C65: the CPU did set N and V flags also for the immediate addressed versions.

Reference: https://discord.com/channels/719326990221574164/782759365064917065/933794977232879627

lydon42, 2022-01-20:
In essence: the MEGA65 behaves like the C65, but not like the 65C02 (which does not set N and V in immediate mode).

Now the decision is still to be made: which one do we follow?

lydon42, 2023-12-13:
The implementation of the MEGA65 copies the implementation of the 4510 CPU of the C65, which does not treat BIT immediate differently then it's addressed variants. The test Bo Zimmermann did confirms this on real hardware.

The Steering Committee confirmed that the current implementation in the core is the correct one, and that we will add documentation to the user-guide to reflect this and avoid further questions.

See also the [65CE02 documentation](https://web.archive.org/web/20221112231057if_/http://archive.6502.org/datasheets/mos_65ce02_mpu.pdf) (the 4510 is based on a 65CE02 core):

> The BIT (IMMEDIATE) test will set the N and V flags
> with valid states, which was not the case with earlier
> 65C02s.

xemu-mega65 will be changed to mirror the behavior of the MEGA65 core.

closed 2023-12-13 by lydon42