MEGA65/mega65-core #530: FREEZER "reset" causes damage to machine state, requiring power cycle to restore

M3wP, 2022-02-02:
When the "reset" function of the FREEZER is used, the machine state becomes damaged, requiring a cold boot to restore.  This can be confirmed with a warm boot and monitoring the boot report for the warning "hicked hypervisor".

Many things will no longer work as expected and especially, using the FREEZER again will mostly likely hang the machine.

There is a related issue:  #499

lydon42, 2022-02-05:
"hicked hypervisor" means that this no cold boot and so HICKUP.M65 is not loaded.

But yes, some stuff stays on a warm boot that better should be reseted... for example the vic-regs that caused #499

lydon42, 2022-09-22:
is this fixed? does someone has a test for this?

closed 2024-08-18 by lydon42