MEGA65/mega65-core #531: Enable mounting of D64 disk images

gardners, 2022-02-05:
We currently support D81 and D65 images. It would be good to support D64 and D71 images.
Bitshifter already has working CBDOS modifications to support these, but the hardware disk image handler needs to know about the new sizes of these images, so that it doesn't allow read or write beyond the end of the disk image.

$D68A bit 6 = D64/D71 enable for disk 0
$D68A bit 7 = D64/D71 enable for disk 1

Combining D64 bit with MEGA Disk bit = 1571 image. So now we have D64, D71, D81 and D65 disk sizes using two bits per drive... once I implement it.

lydon42, 2022-06-13:
Open: HYPPO needs to allow the attaching of D64 images. This currently fails because there is a size check.

gurcei, 2022-06-17:
I've committed a few tweaks to the `d64_support` branch to get it to accept d64 files in hyppo and the freezer.

I can now see the .d64 files in the diskchooser menu, but the disk preview elements within show up as corrupted:
[image]

To my surprise, if I mount some .d64 images while in c64-mode, I can list their directories (not all the time, some d64s work like this, others don't).
[image]

But regardless, I'm unable to load files within the disk at present.

I cherry-picked two of @gardners' commits in relation to #531 that he had on another branch and built a core with them, but still didn't help. 

We can revisit this later when there's time to iron things out further.

dansanderson, 2024-07-01:
Status update: A previous attempt to enable D64 support in the ROM (920381) caused some weird issues in edge cases and was rolled back. I am testing a roll-forward of the ROM change atop ROM beta 920402, and along with core release v0.96, an affected case that could be reproduced in core v0.95 ROM 920381 cannot be reproduced in core v0.96 and a new ROM. We don't have an explanation; my closest guess is something in the core wasn't being reset before and now is, though this doesn't explain why this case appeared to persist state across a cold boot (!). I'm doing extra end-to-end testing on this change, then will be re-releasing the ROM side as a new ROM beta.

> the hardware disk image handler needs to know about the new sizes of these images, so that it doesn't allow read or write beyond the end of the disk image.

Just to confirm, has this part been implemented?

If we're satisfied with the core, Hyppo, and Freezer changes for D64 support, then this issue can be resolved. ROM issue https://github.com/MEGA65/mega65-rom-public/issues/45 tracks the implementation and release of the ROM change.

lydon42, 2024-07-02:
> > the hardware disk image handler needs to know about the new sizes of these images, so that it doesn't allow read or write beyond the end of the disk image.
> 
> Just to confirm, has this part been implemented?

Looking at [99db06f](https://github.com/MEGA65/mega65-core/commit/99db06fb9ad883a6bd8705ffdeed477d55bdf165) I would tend to confirm this:
`
        if f011_d64_disk='1' and (f011_track >34) or (f011_track = 34 and f011_sector > 2) then
          -- Past end of D64 image, point to first sector instead
          diskimage1_offset <= to_unsigned(0,17);
        end if;
`

lydon42, 2024-08-15:
Got some positive feedback from discord with development and 920407, closing.

closed 2024-08-15 by lydon42