MEGA65/mega65-core #536: Add PHQ/PLQ opcodes

lydon42, 2022-02-28:
BitShifter asked if it is possible to implement 32bit push and pop operations to the Q opcode set.
It is used in the ROM and might have some benefit there.

lgblgblgb, 2022-02-28:
The are other situations like that, when having a "Q-opcode" has minimal gain but creates much more complexity (thus may not worth it ...), like having `LDQ #...` (but surely my example is even more worse since it hurts the principle Paul told once, that the length of the opcode should not be altered by prefix, and in this case it would, ie a 32 bit constant after `#` instead of only a 8 bit one). But in theory it's an interesting idea to have `LDQ #...` with a 8 bit constant only, that is, the rest of the "Q" is filled with zeroes, since it's often a case we need a Q-op soon, but the constant would fit into 8 bits quantity (but for this, maybe an opcode for "sign extension" from `A` to `Q` or anything like that would be more useful?)

lydon42, 2024-08-13:
Closed in favor of #645

closed 2024-08-13 by lydon42