MEGA65/mega65-core #537: CIA shift register implementation may be incorrect?

Rhialto, 2022-03-25:
Hi! I'm far from an expert in reading vhdl, but reading this:

https://github.com/MEGA65/mega65-core/blob/c533da5ee6e26d33a6dd5018aac888fd3ece1fcc/src/vhdl/cia6526.vhdl#L549-L554

it looks like the bits are shifted out LSB first. The data sheet however says that "SDR data is shifted out MSB first and serial input data should also appear in this format."

Also, when the processor stores data to the shift register, it's supposed to be double buffered. From the data sheet: "The data in the Serial Data Register will be loaded into the shift register then shift out to the SP pin when a CNT pulse occurs. Data shifted out becomes valid on the falling edge of CNT and remains valid until the next falling edge. After 8 CNT pulses, an interrupt is generated to indicate more data can be sent. If the Serial Data Register was loaded with new information prior to this interrupt, the new data will automatically be loaded into the shift register and transmission will continue. If the microprocessor stays one byte ahead of the shift register, transmission will be continuous.".

This seems to be the code for storing to the SDR but I see no such double-buffering, since the data is written directly into the register that is also shifted:

https://github.com/MEGA65/mega65-core/blob/c533da5ee6e26d33a6dd5018aac888fd3ece1fcc/src/vhdl/cia6526.vhdl#L761-L766

gardners, 2022-03-25:
You are quite likely right, as the shift register stuff has not been really well tested. 
If you are willing to summarise the key points from the data sheet that should be implemented, and suggest what you think would be the right way. Don't worry about it being correct VHDL, just capturing the essence of how it should behave will be helpful.

Rhialto, 2022-03-26:
Well the first part would be relatively simple:
```
 countout <= sdr_bit_alternate; 
 if sdr_bit_alternate='0' then 
   spout <= reg_shift_data(7); 
   reg_sdr_data(7 downto 1) <= reg_sdr_data(6 downto 0); 
   reg_sdr_data(0) <= '0'; 
   report "Shifting out bit, " & integer'image(sdr_bits_remaining-1) & " to go."; 
```
The second one involves an extra register of 8 bits to double-buffer the data, and a single bit to indicate it is filled with valid data. Then, when the shifter is idle, or finished, it moves the data from the new register to the actually shifting register, `reg_shift_data`.
Something like
```
when x"0c" => 
   -- Begin shifting data in or out on shift register 
   if  sdr_bits_remaining == 0 then -- start shifting right away
       report "CIA" & to_hstring(unit) & " Loading shift register";
       reg_shift_data <= std_logic_vector(fastio_wdata); 
       sdr_bits_remaining <= 8; 
       sdr_bit_alternate <= '1'; 
   else -- have to double-buffer this byte
       report "CIA" & to_hstring(unit) & " Loading serial data register";
       reg_sdr_data <= std_logic_vector(fastio_wdata); 
       reg_sdr_filled <= '1';
   end if;
```
and later, for example after `report "Asserting shift register ISR flag";`
```
    if reg_sdr_filled then
       reg_shift_data <= reg_sdr_data;
       sdr_bits_remaining <= 8; 
       sdr_bit_alternate <= '1'; 
       reg_sdr_filled <= '0';
    end if;
```
This is ignoring the weird timing delays that exist around all this...

I'm trying to improve things in VICE, see discussion at https://sourceforge.net/p/vice-emu/patches/233/ and https://sourceforge.net/p/vice-emu/bugs/1219/ for example. There are interesting (read: really annoying) tests for this discussion at https://sourceforge.net/p/vice-emu/code/HEAD/tree/testprogs/CIA/shiftregister/cia-sdr-icr/ .

And there is the shifting IN of course, which I've sort of ignored here.

gardners, 2022-03-26:
Hello,
Thanks for this.  If you are willing to checkout the source, make the changes and then a pull request with them, I'll test that it still builds (unless you want to get all excited and download Vivado etc and test compilation yourself), and then we can merge it into the development branch.

Getting the CIA timing just right is indeed tricky. I believe that Gideon made a harness with a CIA chip on it and FPGA connected to it, so that he could run all manner of tests and see how it behaved.  We could look at doing the same, if someone was enthusiastic at pursuing it.

Paul.

Rhialto, 2022-03-27:
Hi Paul,

see #538 for the pull request. I will leave checking it to you :) I suspect that Vivado is a mswindows-only program and won't run on my NetBSD machine.

I found a CIA in Verilog here: https://github.com/MiSTer-devel/C64_MiSTer/blob/master/rtl/mos6526.v  It says it's not passing VICE's dd0d test; since it's the ICR I suspect that one also doesn't have the timing right, but it may have some ideas in other areas (I didn't really look at it deeply). 

Cheers,
-Olaf

lgblgblgb, 2022-03-28:
@Rhialto 

> I suspect that Vivado is a mswindows-only program and won't run on my NetBSD machine.

Just a comment here, that Vivado also runs on Linux. No idea if NetBSD has some kind of easy & good Linux emulation layer or something like that, but maybe still a better hope as it would be Windows-only.

lydon42, 2022-08-16:
@Rhialto I'd like to take a look at your pull, and I also did take a look at the test suite on sourceforge. Can you explain to me what I want to see with the test programs? I tried one and get a red border screen with filled with 'I' or 'A', toggeling after a press of space.

lydon42, 2022-08-16:
I did build a bitstream with your changes, and the referenced CIA tests still show up mostly red. Some have a some green characters, but in essence I don't see much difference between development head with or without your changes.
So... do you happen to have some test program that shows me that the CIA is now behaving as it should?

Rhialto, 2022-08-16:
Great that you're looking at this!

Unfortunately I don't have the illusion that with my changes, the shift register is perfect. The CIA is just too weird for it. It has various delays in it that don't make any sense. Somebody made a very detailed model of how timers and some other things work, but unfortunately it doesn't include the shift register. Maybe you know it, it is this one: https://ist.uwaterloo.ca/~schepers/MJK/cia6526.html (also available from http://www.zimmers.net/anonftp/pub/cbm/documents/chipdata/6526cia.zip )

So the best I was hoping for here was to make it at least a bit better... and let somebody who knows what they're doing with VHDL take some next step.

Now about the tests. They aren't well documented. Each directory has a readme file and that's it. 

In general, many test programs work in a similar way. They all report a global succes/failure code via some virtial I/O register, so that the testbench.sh script can run all of them headless and print results. This is also reflected in the border colour: red/green. This is for instance the current state of VICE:
```
.../vice-emu/testprogs/testbench$ ./testbench.sh x64sc shiftregister
running tests for x64sc (shiftregister):
Reverted '/mnt/vol1/rhialto/cvs/other/vice/vice-emu/testprogs/drive/rpm/rpm.p64'
gmake[1]: micro64disktool: No such file or directory
gmake[1]: *** [Makefile:38: rpm.p64] Error 127
../CIA/shiftregister/ cia-icr-test2-continues.prg - ok 
../CIA/shiftregister/ cia-icr-test2-oneshot.prg - ok 
../CIA/shiftregister/ cia-icr-test-continues-new.prg - ok 
../CIA/shiftregister/ cia-icr-test-continues-old.prg - ok 
../CIA/shiftregister/ cia-icr-test-oneshot-new.prg - ok 
../CIA/shiftregister/ cia-icr-test-oneshot-old.prg - ok 
../CIA/shiftregister/ cia-sdr-load.prg - ok 
../CIA/shiftregister/ cia-sdr-init.prg - ok 
../CIA/shiftregister/ cia-sdr-delay.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-19.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-39.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-3.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4485-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4485-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4485-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4485-19.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4485-39.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4485-3.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4485-4_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-4_7f.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-generic-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-generic-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-generic-19.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-generic-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-generic-3.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-generic-39.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia1-sdr-icr-generic-4_7f.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-19.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-39.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-3.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-19.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-39.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-3.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-4_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4_7f.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-19.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-3.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-39.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-3.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-4485-19.prg - error 
[... 4 lines cut ...]
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-0.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-0_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-19.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-1_7f.prg - error 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-3.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-39.prg - ok 
../CIA/shiftregister/cia-sdr-icr/ cia2-sdr-icr-generic-4_7f.prg - ok 
[... cut ...]
Most (sub)tests do some set-up, then take a bunch of measurements (typically reading a particular register as fast as possible). The readings are put in the top half of screen memory. If there is something in the bottom half, it shows the expected values (like in VIC20/viavarious). Here the red/green colouring is used per character.

If something is even 1 cycle off, most tests fail. 

I don't think there are many tests that actually look at the contents of the shift register; there seem to be only tests that check if interrupts are generated at the right times. This one is hard enough by itself...

Rhialto, 2024-07-20:
So now that I have actual hardware, I did some experiments. From the VICE testbench, I tried both some fairly basic CIA timer tests (ciavarious) and the cia-sdr-icr tests mentioned above. I tested the C64 core and the GO 64 mode of the M65. Simply because of the work needed to create disk images with the tests I didn't get to do some of the interesting other tests (yet).
I found that the C64 core passes all the ciavarious tests, but the M65 fails all of them (some of the sub-tests pass but for the way the tests are set up that is not good enough).
I was very curious about the cia-sdr-icr tests; in the file C64_MiSTerMEGA65/rtl/mos6526.v I saw no particular hint that the weirdness tested here was taken into account. And indeed, these tests fail on the C64 core (and also on the M65 core).
I think I would recommend to "simply" use the core from the C64 and use it in the M65. That would automatically pass many of the tests. I expect though there will be some work to adapt how the other CIA interfaces to the rest of the M65.

Rhialto, 2024-07-20:
After I looked at the two next to each other,  it looks like more than a little bit of work, more like a rewrite... the M65 version has so much extra stuff. But still.

lydon42, 2024-07-21:
Happy to see that you can now explore this yourself! Sorry that I did not find the time to dig myself into that rabbit hole, too :)

We should keep in mind that the goal here is **not** to be compatible with what the C64 does. So using the C64 cores implementation might be the wrong thing to do.

We need to explore what the C65 prototypes did and implement that! And it needs not to be compatible to the C64, as the C65 really was not...

Rhialto, 2024-07-21:
True, but because there was to be a C64 mode it would make little sense if the CIAs would be not 100% the same, it would reduce the compatibility level (already caused by different CPU etc) even more.

What I remember from reading the C65 description (years ago) was that the CIA description was pretty much lifted from descriptions of the existing 6526.

However you made me doubt everything and I went back to the docs I am aware of, mainly https://www.zimmers.net/anonftp/pub/cbm/manuals/c65/c65manualupdated.txt .

So from section 2.3.3.3. Register Description on the I/O is described. There is a register map that looks like bog standard CIAs, two of them. Except that it is all mashed together, not mentioning 2 CIAs, but timers A-D, ports A-D, shift registers A and B, etc.
Then there's an ACIA and some extra port E and FSERIAL, Fast Serial Bus Control, which is new.

The CIA 6526 datasheet I looked at is https://www.zimmers.net/anonftp/pub/cbm/documents/chipdata/6526.zip.

Then at 2.3.5. Peripheral Control Functions the textual description starts. This very closely follows the 6526 datasheet I mentioned. Except for some weirdnesses.

Port C (CIA 2 port A) is somehow different / incomplete. It's the port where the VIC-II memory bank is generated (this seems to be missing), and the IEC bus interface. That part is reduced somehow from generic I/O bits to IEC-bus specific I/O. A bit weird, but probably the IEC part of this does not reduce the compatibility for C64 mode. The missing VIC-II bank bits seem puzzling. I'm not quite sure what the gain is from this simplification, it seems too minor to be useful for anything.
Apart from this and the port doubling / renaming, the description is (almost) word by word the same as in the datasheet (page 5 and further).
Same for 2.3.5.2. Handshaking.
Same for 2.3.5.3. Interval Timers (Timer A, Timer B, Timer C, Timer D).
It is mentioned that the timers count on the 1 MHz clock (but speed of register access from the CPU isn't mentioned, at least not here)
2.3.5.4. Time of Day Clocks (TODA, TODB) too except for the 10 Hz signal that drives the TOD. In the datasheet, the input is 50 or 60 Hz power line frequency, and the CIA divides it by 5 or 6 (settable) to get the 10 Hz. The C65 seems to use some other clock which is called C7MHz but which apparently can be different actual frequencies depending on PAL or NTSC mode.
2.3.5.5.  Serial Ports (SDRA, SDRB) The same again. It does mention the buffered nature and that "SDR  data  is shifted out MSB first".

And there is some extra hardware support for fast serial mode (2.3.5.6.) How this register is accessed seems contradictory ("The  FAST  SERIAL logic consists of a 2-bit write-only register, which  resides  in  location  0001  (hex)" vs `| 1  0   1  0  0  1  |  2X9  |  FSERIAL  |  Fast Serial Bus Control  |` and `| R/W |2X9|  FSERIAL |*DMODE|*FSDIR|  --  |  --  |  --  |  --  |  --  |  --  |`)

Then there is 2.3.5.7. Interrupt Control Registers (ICRA, ICRB). Curiously the sentence "If bit 7 of the data written is a ONE, any mask bit being written with a one will be set, while those mask bits written with a zero will be unaffected" is missing. Probably just an editorial error.

Then 2.3.5.8.  Control Registers  (CRA, CRB, CRC, CRD) which seems to have some editorial error in "(Bits  0-4  of the CRB register operate identically to bits 0-4 of the
CR7  register," and similarly "(Bits  0-4  of the CRD register operate identically to bits 0-4 of the CRD  register,".

Next up is "6526 cia complex interface adapter \#1" and "6526 cia  complex interface adapter \#2" where we are back to two CIAs, and they are explicitly named.

What port E is for remains unclear. It is just 2 bits. Maybe it somehow is supposed to replace the VIC-II bank bits???

So all in all, so far I don't have reason to think that the CIAs are not meant to be 100% compatible (except for the items mentioned).