MEGA65/mega65-core #542: Remind/enforce YY-MM-DD format in RTC setting in configuration

dansanderson, 2022-05-15:
The configuration screen's RTC setting interface displays the current value of the clock, time and date. The date entry expects the format YY-MM-DD, but does not enforce value ranges. If someone misunderstands the date format and enters MM-DD-YY values, it accepts them, but then some date displays like the BASIC title screen show a corrupted value. The config screen doesn't indicate that there's an issue: it just prints the date as it was entered.

Ideally the config screen would display the expected format alongside the field value, and also enforce ranges. I understand enforcing ranges properly requires deeper logic and UX, but even hard-coded limits of 99, 12, and 31 could be useful. Short of enforcement, a warning message would help.

A "20" before the year value would also be a big win, I think.

On-boarding sets the time using a very different interface. I don't know if any changes are required there.

lydon42, 2022-05-29:
I agree, Date and Time ranges should be enforced by configure.

lydon42, 2022-08-13:
The way time and date is entered and displayed should be unified. This will be decided in the next steering group meeting.

Onboarding uses `HH:MM:SS dd/mmm/yyyy`
Configure uses `HH:MM.SS yy.mm.dd`
ROM displays `dd-mmm-yyyy HH:MM:SS` on startup as well in TI$ and DT$

lydon42, 2022-08-21:
Confirmation by steering group: `HH:MM:SS dd/mmm/yyyy` we will go for and implement in all configuration tools.
In addition range checking should be implemented as much as it is possible. Full leap year day limiting is not a requirement.

closed 2022-09-24 by lydon42