MEGA65/mega65-core #544: Ugly scroll in C64 mode

fredrikr, 2022-05-17:
I ran a small Basic game in C64 mode on the MEGA65 core (the core that was shipped with my MEGA65, serial #311) and the scrolling of the color memory seems to be out of sync with the scrolling of the screen memory, i.e. when the text scrolls up sometimes part of a character flashes in the wrong colour. I tried the same game on a real C64C PAL and it doesn't happen there. Attaching the game.

[torpedo.zip](https://github.com/MEGA65/mega65-core/files/8706550/torpedo.zip)

HovKlan-DH, 2022-05-17:
Did you try to update the ROM to the newest version, 920350? So many ROM changes since the factory default version.

fredrikr, 2022-05-17:
I installed the latest core and the latest ROM. It still happens, but maybe less frequently than before. I also changed my display from HDMI to VGA - could that be a factor?

Edilbert, 2022-05-30:
ROM changes are done for the MEGA65 mode, not the C64 mode.
The C64 mode is not fully compatible with a real C64 for several reasons:
Illegal opcodes do not work as expected, they are now real opcodes for new instructions.
Access to units #8 and #9 is intercepted and handled by CBDOS, not the IEC bus.
Tape routines are removed to make room for bank switching CBDOS calls.

fredrikr, 2022-05-31:
I'm fully aware of that.

I just thought this seemed to be a thing that could be improved in the scrolling routine.

Edilbert, 2022-05-31:
I get already a lot of complains, because I change the MEGA 65 part of the ROM,
which, sometimes inevitable, sometimes by chance introduce incompatibilities.

I'll not touch the ROM part for the C64 mode.
If I would do that, I know, that I would be blamed for every issue, that pops up and there are many already.
But these are currently all heritage of the Commodore 65 developers,
so these are not my business and I can focus on BASIC 65 and CBDOS.

fredrikr, 2022-05-31:
Ok. I certainly didn't mean to complain. I know you've done a great job and you keep at it. Closing this.

closed 2022-05-31 by fredrikr