MEGA65/mega65-core #545: Access a SD mounted disk as DRIVE8 and the internal floppy as DRIVE9 (user configurable drives)

HovKlan-DH, 2022-05-17:
It would be nice to access either a SD mounted disk or the internal floppy, without the need to mount the internal floppy via the freeze. Of course the SD disk must be mounted via freezer, but after that it should be possible to do both of these:

- `LOAD "$",8` (or just `LOAD "$"`) to access the SD mounted disk
- `LOAD "$",9` to access the internal floppy

Both DRIVE8 and DRIVE9 should be possible to configure which real drive it should use (maybe I want DRIVE8 to be the internal floppy and DRIVE9 to be an external floppy).

The configuration should be permanent saved, so the same is set after a power cycle.

cybern0id, 2022-05-25:
@lgblgblgb thanks, understood. I was making assumptions based on observations rather than coming from an understanding of the internals of the C65.

Still, when in Mega C64 mode, both the SD and internal floppy are detected by their internal device IDs as set in the Freezer menu - running the S.T.F.U program shows them as "Foreign" devices and C64 games and demos that demand no other devices on the IEC bus refuse to run. Obviously my interpretation of the inner workings of the C65 are incorrect but it would still be nice to be able to set device IDs for the internal drives, as specified by the OP and also switch internal devices "off". I was imagining that the combined functionality would fit nicely in the Freezer UI.

cybern0id, 2022-05-25:
I should mention that setting the Freezer menu so each of the devices are set to "No Disk" still results in the device IDs being detected in C64 mode. Perhaps I do need to raise a separate github issue / feature request after all?

lgblgblgb, 2022-05-25:
I think, the problem here that part of this story is up to the ROM ... For example the hardware of MEGA65 itself can say what is the internal drive 0 and 1, ie internal disk drive, or "D81 mount". There are (often called drive 8 and 9 ... But for real that "8" and "9" is already not by the hardware so the freezer has no control over it, it's the ROM. And btw the previous topic, even on C64, device number 1 is TAPE, thus just by having some device number does not mean it must be the IEC bus at all, it's just the notion of ROM nothing more or less. Freezer can just assign what medium is treated as internal drives (real internal floppy, or a D81 "mounted") but cannot rewrite the ROM who has the control what device number assigned to what hardware devices let it be IEC bus, FDC internal drive, or even TAPE (not on MEGA65 but eg on C64) or even screen ....

oe3gwu, 2022-05-29:
It would be nice. Just formatted my mega65.d81 because it defaulted back.

Edilbert, 2022-05-30:
That's why the routine asks: ARE YOU SURE?