MEGA65/mega65-core #551: A formatted real disk on MEGA65 is not readable on CMD's FD2000 or FD4000

Spockie32, 2022-05-26:
**Test Environment (required)**
 - Platform: MEGA65 R3 (Retail)
 - Core Commit: bdeeb15
 - ROM Release: V920353

**Describe the bug**
If you format a real 3.5" disk with the MEGA65 in 1581 format (with the HEADER command), CMD's FD 2000 and FD 4000 can't read it, although both drives are compatible with the 1581. They produce a "Drive not ready" error. The other way around works: So on a CMD FD formated disk (in 1581 format) is readable on the MEGA65. 

**To Reproduce**
Steps to reproduce the behavior:
1. In the MEGA65 freezer switch to the internal 3.5" drive.
2. Insert a blank DD disk.
3. Format it with HEADER "MEGA65 DISK",I65
4. Answer the question (Are you sure?) with Y
5. Wait to finish formatting. Check if it worked with a simple DIR.
6. The MEGA65 reports back a normal directory with 3160 blocks free
7. Take that disk and insert it on a FD 2000 or FD 4000 drive on a C64
8. Try to load a directory
9. The C64 gives you a "Drive not ready" error.
10. Take that same disk and try to read it on a 1581. There it works.

**Expected behavior**
The disk should be read- and writeable on the FD 2000/4000 just as it is on a 1581.

**Additional context**
The other way around works: 
Insert a disk into a CMD FD 2000 or 4000. 
Under JiffyDOS enter "@n:mega65 disk,65" and wait the format process to finish. 
You end up with a 1581 formatted disk with 3160 blocks free. 
Insert this disk into a MEGA65 and it works perfectly.

lgblgblgb, 2022-05-26:
_Only_ an idea: have you covered the HD hole on the disk if you use HD disks? I can imagine it's a problem that MEGA65 for real uses (I guess originally from a PC) HD drive. Not so much ida on FD 2000/4000 though. Especially strange that 1581 can read it but not FD 2000/4000. This means even if it's stated FD 2000/4000 is 1581 compatible certainly not fully otherwise those wouldn't behave differently, I guess.

Spockie32, 2022-05-26:
Hi, thanks for your thoughts. I've used a regular DD (Double Sided, Double Density) disk, to rule possible issues with an HD disk out. So that's not the problem here.

I know that the FD 2000/4000 are not 100% compatible to the 1581, but a simple formated disk should be readable.

oe3gwu, 2022-06-01:
I have formatted a DD and a HD Disk with my MEGA65 in 1581 format and it works well on my Firedrive 2k. Also a Firedrive 2k formatted 1581 was readable from my MEGA65. I didnt do extensive testing though. Firedrive 2k = CMD FD-2000 Clone

Spockie32, 2022-06-01:
The RearAdmiral Firedrive is obviously not a 100% clone of CMD's FD drives. I've just tried it again with my CMD FD 4000 and the same error persists: A 1581 formatted disk on MEGA65 is not readable on a CMD FD 4000. A friend of mine tried the same with his FD 2000 with the same result. The same disk works on a regular 1581 however.

dansanderson, 2024-07-01:
fwiw I just noticed this as well. Nice to see there's already an issue created. I have an FD-2000 and an FNX1592HD (same result) I can test with, if/when anyone looks at this issue.

lydon42, 2024-09-01:
Confirmed by @dansanderson , so we make this a bug.