MEGA65/mega65-core #552: Hyppo / Config run in HDMI mode even if DVI only is configured

lydon42, 2022-05-28:
Reported by @dananderson as https://github.com/MEGA65/mega65-rom-public/issues/4

**Test Environment (required)**
 - Platform: MEGA65
 - ROM Release: 920355
 - Core Commit: bdeeb15

**Describe the bug**
The "Enhanced" (sound enabled) DVI video mode does not work with DVI-only displays, which is why it is an option in the configuration. However, it appears that several of the pre-config menus, including the Hypervisor, core menu, and the config menu itself use "Enhanced" mode. This breaks these screens on DVI-only displays, and prevents the user from accessing the config menu to change it for the rest of the machine.

As I understand it, DVI-only mode should show a correct picture on all displays using the DVI output, including modern HDMI displays that otherwise support sound. Hypervisor et al should use DVI-only mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a DVI-only display. I'm using a Dell 2001FP and an HDMI-to-DVI video cable.
2. Hold Alt, then turn on the MEGA65.

**Expected behavior**
The Hypervisor Alt menu should display correctly. Instead, it behaves as if in "Enhanced" video mode.

The only workaround I know is to connect a VGA display to access these menus. It so happens that the Dell 2001FP has both VGA and DVI inputs, so I keep both connected and switch between them as needed. I now realize that this would not be necessary if these menus used DVI-only mode.

lydon42, 2022-05-29:
Someone needs to test this, my DVI monitor can't handle the MEGA65 signal.

dansanderson, 2022-05-29:
If there's a candidate fix I'm happy to test an experimental core where I can reproduce this issue.

cwscws, 2022-06-06:
Would be very much interested in a fix. Using VGA at the moment but would prefer DVI. Unfortunately black screen when in config/core select via DVI.

Would a fix be applied with a Mega65 core update?

lydon42, 2022-06-06:
I can reproduce it, probably happens because config is read after start of mega65_config.prg

lydon42, 2022-06-06:
Try the following (works best if you also have VGA connected to not have to write blindly):
```
Boot with ALT, press 1 for configure.
MEGA-TAB
s777d61a 02
```
now DVI should show a picture.

I can't fix it with a HICKUP, as it seems, as the startup stuff all comes from unmodified core in slot 0. And fixing mega65_config will also require reflashing of slot 0. So there is probably no workaround without flashing slot 0.

Problem is hyppo/main.asm - safe_video_mode, which sets $d61a to 0 instead of 2. Possible fix is also to add a sta of 2 to §d61a in init of mega65_config.asm.

dansanderson, 2022-06-06:
I can confirm that s777d61a 02 restores a configuration screen on the DVI display.

Are there plans to ship batch #2 machines with an updated core 0? In theory this bug locks out users with DVI-only displays, but I don't know how big that user base is, and in practice most DVI-only displays also have VGA analog inputs and we can coach those users toward the workaround of using both inputs. I'm more inclined to recommend "use a VGA or HDMI display for config" as the workaround than "get a special difficult-to-find cable and carefully overwrite core 0." If we have the opportunity to prepare a new factory core for batch #2, I'd prioritize this fix.

dansanderson, 2022-06-06:
P.S. I do have a full JTAG adapter and I'm willing to try a new core 0 to test. I'd need instructions.

lydon42, 2022-06-07:
fix core created, testing with @dansanderson

dansanderson, 2022-06-07:
Testing successful. The fixed core runs boot menus in DVI-only video mode.

The fixed core must be in slot 0 to run the core selection menu in DVI-only mode. We verified this by first testing the fixed core in slot 1 (ALT utilities worked, NO SCROLL core selection still did not), then flashing it to slot 0 via JTAG (both ALT and NO SCROLL utilities work).

cwscws, 2022-06-07:
Great! But have no JTAG adapter. And it was said that nobody should mess with slot 0...

lydon42, 2022-06-07:
@cwscws  flashing slot 0 is possible without JTAG. It is just more dangerous, because if the flashing goes wrong, you will have no way to get the MEGA running again. So we advice against this. We will be discussing this topic in an upcoming meeting.

cwscws, 2022-06-07:
I think it would be better to wait for a more stable core before flashing to slot 0. VGA works anyway, DVI would just be a nice bonus.

closed 2022-06-07 by lydon42