MEGA65/mega65-core #553: CMD swap button compatibility

oe3gwu, 2022-05-29:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 7C5281E
 - ROM Release: (i.e. 920357)

**Describe the bug**
CMD FD-2000 and CMD-HD Swap Buttons dont work properly. This button swaps the CMD Jumpered Drive Number with the internal drive 8. It works on real hardware, aswell aus the Ultimate 64. However, if you press the button on the MEGA65, nothing happens. A workaround is, that you can reconfigure the internal drive to an unasigned number, then the swap button works. 

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the MEGA65
2. Click the swap button
3. Swap LED does NOT light up.

**Expected behavior**
The internal drive 8 and the jumpered drive letter from the CMD drive should swap positions.

lydon42, 2022-05-29:
Is this related to #554 ?
Does the drive itself work?

oe3gwu, 2022-05-29:
No I use damaged drives all time time. Of course they work and work perfect on my Ultimate 64 Elite.

gardners, 2022-05-30:
The reason that this doesn't work, is that the CMD swap button works by breaking the IEC standard, by temporarily becoming the bus master -- which only the computer is allowed to do -- and then connecting to whatever drive is on the other device ID, and instructing it to change it's device number, and then the CMD device changes its device number internally as well.

This never worked on any of the Commodore 65 prototypes because the C65's internal drive is _not on the IEC bus_.  The C65 KERNAL doesn't listen for IEC commands from another bus master.  In fact, it _can't_ listen for IEC commands from another bus master, because the ATN line is output only on the C65 (and the C64, for that matter).  So there is no way for a C65 to know that the CMD device is trying to take over the world temporarily (for the purposes of good of course ;).

Even if the C65 were physically able to watch for this situation, it would only work while the KERNAL was active.  It would mean that the KERNAL would have to be running an interrupt _very_ frequently to watch for the ATN line mysteriously going low.  The instant you were running software that disabled the KERNAL and used its own interrupt routines (i.e., a very large fraction of all software), it would also cease to work.

The MEGA65 has exactly the same function here as a C65, and thus the CMD drive swap button will only work with IEC connected drives. The MEGA65's internal drive is, just like in the original C65, not on the IEC bus, and so all of the above discussion still applies. 

The only way to fix this would be to make the MEGA65 do the drive swap function, instead of the CMD drive breaking the IEC protocol to do it.  

The FD2000 manual (eg from ftp://cbm8bit.com/hardware/manuals/CMD-FD2000-Manual.pdf) confirms that this is how the drive swap button works (top of page 15):

"SWAP can be used even if you do not already have a device 8 or 9 on the serial bus. In this case, the FD simply becomes device 8 or 9 without affecting any other devices on the bus."

It's a little vague, but the clear implication of what they are saying is that if the FD2000 doesn't find another device on the IEC serial bus, then it will just switch to the swapped device number.  This is exactly what happens with a C65 or a MEGA65 (or for that matter, a C64 with an REU ram disk on device 8, or anything else on device 8 that is not on the IEC serial bus).  

The key give-away here is "without affecting any other device on the bus" -> The FD2000 drive swap can by its own admission only affect other devices on the IEC serial bus. The C65 and MEGA65's internal drive is not on the IEC serial bus, but is rather directly on the processor's address and data bus, and it appears on device 8 because the KERNAL intercepts accesses to that device number before they even hit the IEC bus -- just like an REU ram disk would.

So as much as it would be nice to support the CMD drive swap buttons, this is not physically possible.  The work-around is to use the freezer or other means to shove the internal drive off of device 8.  It would also be possible to implement a function in the freezer that does the swap. You are welcome to implement this, as I am sure a number of people would appreciate it.

But as for the original issue, it is not possible.

closed 2022-05-30 by gardners