MEGA65/mega65-core #554: 1581 Game Compatibility

oe3gwu, 2022-05-29:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 7C5281E
 - ROM Release: 920357

**Describe the bug**
There is a website for download converted games to 1581 (D81) format. However, 2/3 of the games stall in loading. Like Test Drive II. After detecting the drive, nothing happens - just before the Accolade Logo. Be aware, the icons on the website dont work anymore, but the download links are still valid.

I got those images on CMD-2000 FDD's. Due to a Freezer Bug, I cant mount the game via the C64 Freezer Menu.

The Disks work perfect using a real C64 or a Ultimate 1541 Elite.

http://members.optusnet.com.au/spacetaxi64/MAIN/1581-GAMES.htm

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the MEGA65
2. Mount the Test Drive II D81
3. Load the game
4. Change cheat options as you want
5. N0S Loader will detect the Image (od Drive)
6. Loading stalls

**Expected behavior**
Game should work

gurcei, 2022-05-29:
I've also found slme c64 releases a bit problematic to load up. I do recall someone saying to try choose versions of software released by groups like REMEMBER, as they didn't use any illegal instructions in their code. I wodner if that will help (I haven't tried on my end).

If illegal instructions are a necessity though, they are available via the pure C64 core.

lydon42, 2022-05-29:
Please clarify: are we talking about real disks that are read in some IEC attached drive? Or real disks written with this CMD-2000 and then read in the internal drive?
Or is this a problem that the specific game does not work? That's no problem, it is expected that not all C64 games will work in C64 mode.
If we are talking IEC device problems, that could be an core issue.

oe3gwu, 2022-05-29:
Real disks, I use a Rear Admiral FireDrive 2k.