MEGA65/mega65-core #555: REU Builtin

oe3gwu, 2022-05-29:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
D81 Images sometimes tend to load them into RAM. This is true for Sonic the Hedgehog for example. So a REU Support should be added. But dont get me wrong, I am not willed to use a REAL REU Brick on an FPGA Computer. It would be nice to have it built in.

**Describe alternatives you've considered**
I tried to use my real REU Cartridge, but it doesnt work. I am not developer enough to debug this, however the commands that are added from the REU to Store and Fetch Data arent working either.

lgblgblgb, 2022-05-29:
@oe3gwu REU is a C64 "thing". So you can't expect too much to be in the mega65-core. It is more suited for the C64 core, but this "issues" section is for mega65-core, not the C64-core (also called "alternative cores", C64-core being one of them). The MEGA65 style solution for this problem is to fill the hyperRAM ("Attic RAM") which is 8Mbyte in size (you can use even BASIC65's `BLOAD` - I believe - to load things there). MEGA65-core's C64 compatibility is limited by design as C65's one too (and MEGA65 based on the C65 design). That is the reason for better C64 compatibility, a dedicated C64 core should be used, it's a common mistake to think that MEGA65-core should be 100% C64 compatible (even then' MEGA65's "C64 mode" is way more C64 compatible than C65's, but because of reasons I mentioned it won't be ever 100% and it's not even the goal, that's more a C64-core territory). Otherwise of course it wouldn't hurt to have REU in the MEGA65-core as well, however the problem that it's impossible to do: mega65-core internal memory decoding and its timing is so much overloaded, that it's about virtually impossible to try to "force in" even more things in this respect, at least currently.

oe3gwu, 2022-05-29:
Well it is a MEGA65. The C64 core is builtin the MEGA65.ROM, so this is a MEGA65 issue. The REU Units can also be used for the C128. So why not the M65?

lgblgblgb, 2022-05-29:
Oh, no, there is a big misunderstanding here. C64 core is not builtin the ROM! The core itself what configures the FPGA to use routings between its logic blocks, LUTs etc to routed in a given way, ie a creating a "hardware circuit" inside the FPGA using its resources. That is what mega65-core's or the C64-core. The ROM is only a software running on top of the hardware (in our case, on the hardware created inside the FPGA). MEGA65's "C64 mode"  you see after "GO64" is nothing to do with the C64-core! It's a C65 "legacy" and it just does that the MEGA65-core and its created hardware gets settings in I/O register to try to mimic the defaults of a C64 more or less (like, setting VIC-IV to have 40 column display etc, but it's still a VIC-IV and not VIC-II! So won't be even _ever_ 100% C64 compatible just more or less ...). At the other hand, C64-core is a core, which sets up the FPGA to implement a totally different internal logic for implementing a C64 computer. Cores needs to reconfigure the FPGA. Thus, for having a simple example, core is like having a C64, C16, Plus4 etc so the hardware itself (just in MEGA65's FPGA case, it can be altered since it's an FPGA not a fixed PCB with fixed chips like a real C64 or C16, etc ...) while ROM is only a software running on the hardware, it cannot be a "core" since it's software and not hardware (I mean the ROM content, for sure in a C64 ROM itself is also a chip ... I didn't meant that).

Hopefully it helped to cure the misunderstanding here what a core is. Even with my bad English :) C64-core is a totally different entity, with its own core files needed to be uploaded to the FPGA to re-create a C64 not a MEGA65 hardware inside the FPGA.

oe3gwu, 2022-05-29:
Well, I am no developer. As User I just would like a REU Support. I need it for many operations on the C128 too. This is a basic function that EVERY other FPGA C64 has got.

If this is thr wrong place, then tell me where to put it.

lgblgblgb, 2022-05-29:
Yes, absolutely, I will discuss this "wish place" idea with the others which is not mega65-core topic strictly. However the other key factor here that you wrote: "This is a basic function that EVERY other FPGA C64 has got". Well mega65-core is not a c64 core. That would the the alternative C64-core but not the mega65-core. So mega65-core is not needed to be C64 compatible at all. As I've said even the "Go64" is still not the C64-core, it's a C65-legacy (you can say mega65-core is a C65 - not C64 - core though with large amount of own inventions meanwhile to become the MEGA65 now) to provide a very basic level of C64-like environment more or less, without real "strict" C64 compatibility. For a better C64 compatibility, surely another FPGA core should be used, like the C64-core I mentioned now several times, not the mega65 FPGA core.

oe3gwu, 2022-05-29:
Well if it is no C64 and I shouldnt have any hopes that C64 functions are added, you should remove it entire. I mean I am very frustrated now. I gonna shelf this project. If nothing changes the next month, I just sell it. I am out, completly!

lydon42, 2022-05-29:
This is not a MEGA65 issue. The C64 core has REU on the roadmap.

closed 2022-05-29 by lydon42

gardners, 2022-05-30:
Also, it must be said, that there is already partial implementation of the REU in the MEGA65. It may well get completed at some point, but requires someone with the time and energy to tackle it.  You are welcome to contribute to this. I will provide support, but it will require someone who knows (or is willing to learn) VHDL.

Now, in terms of removing the "GO64" and the C64 mode of the MEGA65 -- this makes no sense: What we have, is exactly what the C65 had in this regard, a _partially_ compatible C64 mode for _partial_ compatibility with existing software and _some_ hardware. The C65 Specifications Manual document from Commodore is quite explicit about this. I think they call it "moderate compatibility" with hardware.  In fact, you couldn't even _physically fit_ a C64 REU in the C65's cartridge port, because the C65 had a Plus/4 style cartridge port instead. So we have already improved the hardware compatibility with the C64 quite a lot!

A more fruitful approach would be to patch any software that expects an REU to instead use the MEGA65's DMA controller and Attic RAM.  This would not be particularly hard to do for most titles.

gardners, 2022-05-30:
Also, please remember that we have been totally up-front that the MEGA65 is a _community work in progress_. It gets better by everyone contributing. If you were looking for a machine that was finished from day 1, then I am sorry, the MEGA65 is probably not the machine you were looking for.  If on the other hand you have patience, or the desire to learn how to solve some of these rough edges, then you will have a great time along with the rest of us. But that is not a decision we can make for you.