MEGA65/mega65-core #556: Kung Fu Flash Cartridge doesnt work correct

oe3gwu, 2022-05-29:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 7C5281E
 - ROM Release: 920357

**Describe the bug**
Kung Fu Flash is a software defined cartridge. The last configured CRT or PRG will automatically start. This works fine. However, if you want to reset the MEGA65 through the reset button, a blue screen appears. If you want to go to the freezer menu and load an other software, the screen scrambles and the MEGA65 hangs. Again works perfect on the Ultimate 1541 Elite.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the MEGA65
2. Click the Menu or Reset Button on the Kung Fu Flash

**Expected behavior**
Freezer and Reset should work.

**Screenshots**
[image]
[image]

**Additional context**
If I get my cartridge back, I can ship it to a developer for debugging.

lydon42, 2022-05-29:
This is a #564

Schwefelholz, 2024-08-18:
On my R3A using cores RC fb7e8f0 and 0.96 Release, I cannot get my Kung Fu Flash to run at all.
I just get the second clean blue picture as shown in https://github.com/MEGA65/mega65-core/issues/556#issue-1251851024

nobruinfo, 2024-08-18:
On R6 and da9b0fb and the earlier 0.96 3c10488 release core this just works for me.
[image]
[image]

https://github.com/user-attachments/assets/6e7295e9-ce3c-4f12-988a-f0e3a677f5c9

lydon42, 2024-08-18:
> On R6 and [da9b0fb](https://github.com/MEGA65/mega65-core/commit/da9b0fb8d02b224eeeafcca18f7049a09a60c434) and the earlier 0.96 [3c10488](https://github.com/MEGA65/mega65-core/commit/3c1048830bb6f543cb832049e969301a9a389fa2) release core this just works for me.

Just to clarify: the Cartridge does run in the C64 Core not in the MEGA65 core. But detection through MEGAFLASH works well, as the C64 core seems to be started automatically.

Pressing HELP will bring up the C64 cores menu, thats a clear indicator (but you did not press it in your video).

nobruinfo, 2024-08-18:
Yes, I didn't think of the [HELP] key.
[image]

paich64, 2024-08-18:
Kung FU flash observations on **R6**:

**With the C64 core V5.1** : It perfectly works, all buttons are working as expected.

**Without the C64 core** (i.e using the Mega65 core) : 

- If a game was previously saved as the game to start (operation done when used with the C64 core), Then the cartridge the Mega65 core will boot the cartridge and launch the game.

- Now having the game launched, MENU & RESET button won't work correctly :

While a game is running, MENU will not come up when hitting the "MENU" button. instead the game freezes and the left/right borders are flashing really fast.

The only workaround to access the MENU (while having the game frozen and left/right borders flashing) is to reset the MEGA65 with the Mega65 reset button. 

As soon as the Mega65 resets, it boots the cartridge and display the KFF menu.

It's possible to navigate in the menu, but trying to load a game will fail [black screen] (Normally the KFF flash cartridge is supposed to load the game in memory and start it) 

- The RESET button won't work : instead of resetting, it will freeze the game. Any attempt to reset the Mega65 with the Mega65 reset button will result in the Mega65 resetting but not booting the cartridge.

Instead we get to the Basic 65 prompt. We need to power cycle the Mega65 to having it booting the KFF again and the Game (already launched when using the C64 core) will start immediately.

What it means : It means that **on R6 with the MEGA65 core, the KFF partially works but it barely unusable since it's not possible to launch a game.**

Kung FU flash observations on **R3/R3A**:

**With the C64 core V5.1** on R3A, It's already known as not capable of working as expected because of the Cartridge Lines limitations in R3A

Read https://github.com/MJoergen/C64MEGA65/blob/master/doc/cartridges.md in which the is a section which clearly states :  

_The circuits of the MEGA65's R3 and R3A boards are not able to detect the KFF's signals properly. Therefore you need a workaround to work with the KFF. It takes a bit of practice to get the timing right, but as soon as you are getting the hang of it, you can work very well with the KFF and the C64 for MEGA65 core_

Personally i can't experience on R3A since i have a problem with my cartridge port. Even the explained workaround won't work for me. 

**Without the C64 core V5.1**

For the same reason I can't experience on R3A

lydon42, 2024-08-18:
So the todo here would be: get Kung Fu flash working on r6 with the MEGA65 core. Possibly so that a M65 style CRT can be used with it.

Schwefelholz, 2024-08-18:
> So the todo here would be: get Kung Fu flash working on r6 with the MEGA65 core. Possibly so that a M65 style CRT can be used with it.

From my perspective it would be better to use an EF3 instead of a KFF for that, because I actually can boot an EF3 with the M65 core on as R3A...
[image]