MEGA65/mega65-core #557: Epyx Fastload Cartridge doesnt work

oe3gwu, 2022-05-29:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 7C5281E
 - ROM Release: 920357

**Describe the bug**
When booting up with the EPYX Fastload Cartridge Reloaded, the computer doesnt promt the normal basic screen. Only FASTLOAD and READY appears. This would be OK, but loading a game doesnt work either. Only Ready appears, nothing else happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Insert the Cartridge
2. Turn on the MEGA65
3. Review the screen.

**Expected behavior**
The Epyx Fastload Cartridge FASTLOAD should appear below the normal BASIC screen and should work.

**Screenshots**
[image]

**Additional context**
I can send the Cartridge to a developer, but I would like to get it back. The internal boot routines are REALLY slow, so some FASTLOAD would be nice.

lydon42, 2022-05-29:
This is a #564

lydon42, 2024-08-18:
The cartridge is reported to work on r3 with the C64 core, which is good, as this is a C64 cartridge.

There is no huge priority to support this on the MEGA65 core.