MEGA65/mega65-core #558: Add support to upload different ROMs like JiffyDOS, JaffyDOS, SpeedDOS, ExOS, ...

oe3gwu, 2022-05-29:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would like to use a JiffyDOS ROM on the C64 Side. It would be nice to have to one big MEGA65.ROM but different ROM's. One for the MEGA65, and one for the C64 Side.

**Describe alternatives you've considered**
The real C64 ROM is just slow to load stuff and annoys me now the same way as in the past.

lgblgblgb, 2022-05-29:
@oe3gwu So as far as I can see, it's more like C64-core request. MEGA65's C64 mode anyway uses CBDOS thus for the internal drive (and for "D81 mount") it's nothing to do with the IEC bus, these solutions won't even work (as these "fastloader" solutions are for speeding up the IEC bus communication between the computer and the drive on the IEC bus). MEGA65-core is based on the C65 design thus not using IEC bus at all for its internal drive access + D81 "mounting".

For sure, in theory you can you can use an IEC bus drive with the MEGA65 + mega65-core, indeed, when it can be useful, that's true. But having a custom C64 ROM in "C64 mode" of mega65-core would lose all possibility to access the internal drive and SD-card D81s from C64 mode. Thus it's more a sane solution for the C64-core, in my opinion at least.

lydon42, 2022-05-29:
The C64 core will be able to load other ROMs, as the MiSTer can, it is on the roadmap.
I am unsure how this will translate to the MEGA65 core. We need to discuss this.

Edilbert, 2022-05-30:
The C64 mode of the MEGA65 uses CBDOS for the internal drive and the SD-Card images.
This is way faster, than any IEC bus solution, like Jiffy-DOS and others.
Alternative ROMs like JiffyDOS don't know about CBDOS. The internal floppy drive and SD Card images couldn't be used with such ROMs.

gardners, 2022-05-30:
Also don't forget that you can already use a C64 JiffyDOS ROM on the MEGA65 -- its just that it will then only boot into C64 mode, and will not be able to see the internal 3.5" drive or SD card or any other C65 or MEGA65 feature anymore, as the ROM will be managing the machine as though it were a C64. It _will_ however make your external drives load at JiffyDOS speed.  You can have multiple ROMs on the SD card named MEGA651.ROM through MEGA659.ROM, and use those on a boot-by-boot basis by holding down the number keys 1 through 9 when resetting the MEGA65. Thus you can easily switch between the normal MEGA65 ROM and the JiffyDOS C64 ROM based on what you want to do at the time.

In the alternative, you are welcome to offer assistance at completing the JiffyDOS support in the MEGA65's ROM.

oe3gwu, 2022-05-31:
Ok I understand!

I have to try this. I wasnt aware, that different ROM's were already compatible already with the MEGA65.

Thank you!

Mikey351AU, 2023-01-27:
> Also don't forget that you can already use a C64 JiffyDOS ROM on the MEGA65 -- its just that it will then only boot into C64 mode, and will not be able to see the internal 3.5" drive or SD card or [...]

Hi Paul, 

What are the requirements to create a MEGA65x.ROM file for a Jiffy C64 mode. I guess it will need to contain basic+jiffykernal+char rom ? 

Mike

dansanderson, 2023-02-02:
MEGA65.ROM is an uncompressed binary file that gets loaded into 3.000-4.FFFF by the Hypervisor. The segments of the ROM follow the C65 layout documented in the manual, with DOS and C64-mode ROM regions in the first half. Registered MEGA65 owners can get read access to the Closed ROM source repo (MEGA65/mega65-rom) if you want to look at the source. You can modify it, build it, and generate binary diffs for distribution. (The complete ROM cannot be distributed freely because it contains licensed material. End users can reconstruct the ROM with the M65Connect tool and the official C65 ROM distributed for free by Cloanto.)

Alternatively you can prepare it as a runtime patch with a loader routine that disables the Hypervisor write protect. It might be challenging to keep it up to date with future ROM changes, but runtime patching might be an interesting way to offer it so it can be something that an end user can just load from a boot disk.

closed 2024-09-10 by lydon42