MEGA65/mega65-core #560: MAC Address is empty

oe3gwu, 2022-05-29:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 7C5281E
 - ROM Release: 920357

**Describe the bug**
Since updating to the new Core (183.0), the MAC Address isnt listened anymore in the Config Menu. I can add one, but it should read out correct from the hardware. I have no idea if it really tries to connect with no mac somewhere.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the MEGA65
3. Boot in Config Menu

**Expected behavior**
Correct view of the MAC.

**Screenshots**
[image]
[image]

lydon42, 2022-05-29:
I can't reproduce this. Are you sure that you had a non-zero MAC before the core upgrade?

But I remember that it is possible that the MAC gets zeroed out. That is no big problem, just set a new MAC (local unicast - x2‑xx‑xx‑xx‑xx‑xx) and you are good to go. The MAC is a random number anyways.

Perhaps this could also be a function of the config menu to set a new pseudo random MAC. I somehow remember this could be something that the onboarding does...

oe3gwu, 2022-05-29:
I am sure I had a MAC in there before!

gardners, 2022-05-31:
Any network software for the MEGA65 should check the MAC address stored is valid, and if not, it should use the MIIC interface to extract the MAC address of the phy hardware.  So, yes, it would be good to fix this.  The configure program is somewhat size limited, so it may or may not be possible to fit it in there.  You are welcome to take a look and propose a patch that fixes it, and if it fits and does the job, we can accept it.

cybern0id, 2022-06-06:
I had my MAC address reset to zero also (no idea how unfortunately. It was definitely not like that at onboarding). However, if you go to the "Done" menu in the configuration app, then choose the "Reset to factory defaults" option and save, you should find the MAC address has been restored when opening the configuration app again.

lydon42, 2022-10-02:
MAC randomizer was added to configure, can be used by pressing R in the MAC field. Rest to Factory defaults will also randomize.

closed 2022-10-02 by lydon42