MEGA65/mega65-core #561: (S)FTP Client builtin

oe3gwu, 2022-05-29:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would love to have an FTP Client to native connect to the Filehost and get online COR and Kernel Updates. Also maybe connect to 3rd party FTP for File Download.

**Describe alternatives you've considered**
I currently do this via a PC: It works, but still ...

**Additional context**
None.

lgblgblgb, 2022-05-29:
@oe3gwu As with #562 this is a software not so much a mega65-core idea/bug itself. This one would be indeed very useful and even planned in theory, however the same issue as with #562 that the team is more focusing now with the hardware platform itself, and the mega65-core, and even "only" this totally overloading everybody. However as I wrote at the other issue as well, MEGA65 is a community effort, so of course we would be very happy to take this job, let it be you, or anybody else. Surely maybe in the future of "more relaxed time" it's quite possible that even team members will have more time to do things like this as well. Of course personally I would like to do things like this (as well). However time is always the limiting effort, as it's just a non-profit hobby for all of us. And I totally agree, it would be a very cool feature not to need to have a PC for this task, but doing it on the MEGA65 itself only :D :D

oe3gwu, 2022-05-29:
I understand that this has no priority. :-)

lgblgblgb, 2022-05-29:
Btw, I really think, we should create a software-idea kind of bug tracking / wish-list place as well, at this point :D Do not misunderstand me, all ideas counts and useful (and should be remembered), just maybe would be better to have the correct place for those, much easier to have an overview and even making people's life easier who want to contribute to an idea with some actual work. But this _my_ opinion now, not the team's common one, so stay tuned please.

oe3gwu, 2022-05-29:
Thats a good idea.

lydon42, 2022-05-29:
Again, no core related feature, but a an application. Probably to be integrated into the freezer or as an standalone option.
There are already thoughts going around (@gardners ) implementing an ethernet based alternative to the JTAG interface. But again this does not need additional support from the core.

Yes, we will thing about a better way to create overarching feature requests that can't be easily assigned to on specific MEGA65 repository. And it's not helping that we can't move issues...

closed 2022-05-29 by lydon42

gurcei, 2022-05-29:
Understanf this is off-topic to mega65-core, but still thought it was worth a mention that maybe the mega65_ftp tool can help you scratch that itch a little bit. While it is still a pc-based tool, it does have commands like `fh` and `fhget` that retrieve files directly from the filehost and transfer them onto your sd-card (via the serial/jtag connector). If you'd like more info/details on this, happy to talk it over further on the discord chat server.