MEGA65/mega65-core #562: CCGMS Mega65 Ecition

oe3gwu, 2022-05-29:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
The MEGA65 got Ethernet. Would be nice to have a CCGMS Edition that can use this Port. Like the CCGMS Ultimate 64 Edition.

**Describe alternatives you've considered**
No

**Additional context**
No

lgblgblgb, 2022-05-29:
@oe3gwu So CCGMS is a BBS software? Then this issue is a bit pointless to be _here_, since it's the mega65-core repository and its bug tracking: that is, about bugs, wishes, etc of the MEGA65 core itself. Writing a software which runs on MEGA65 is out of scope of _this_ repository.  Anyway, of course the idea is indeed nice, and thanks for your idea, however it's nothing to do too much with mega65-core itself. Certainly every software written for MEGA65 is warmly welcomed, but someone should do it. The team is already overloaded with the tasks about the MEGA65 hardware, core, its internal "system" software (like HYPPO, Freezer, etc) and tasks like that (so about the platform itself), thus it's kinda unlikely we can do what you would like, but of course, go ahead and do it, very nice idea, and we would be very grateful to make MEGA65's software portfolio larger.

oe3gwu, 2022-05-29:
Yes and no.

The U64 variant came out after requesting it to the the U64 Developer Gideon. The MEGA65 has got good hardware. It should get in touch with the Devs of other Software. If I request it, I am just a small person.

lgblgblgb, 2022-05-29:
The problem here, that U64 is a kinda well established C64-like platform for now with some past already, also focusing on being a C64 platform, which is of course a very "old" and known platform. Thus - I suppose, surely I can't know - Gideon may have more time and ability to also work with not-strictly U64 related work, but connected: creating/supporting software for his platform. However at least currently, it's not so much the same for MEGA65 and especially MEGA65-core,  also the focus is different, not C64 but C65 based platform (the mega65-core I mean, as a core), also creating its very own new platform, since MEGA65 is having so much new features that the "C65-based" statement is even a bit "understatement" for now already ;)

oe3gwu, 2022-05-29:
Well I do have a fundamental question.

Why do we discuss such "problems"? The M65 has got a ETH Jack. So it is more then logical to ask for that kind of support. ALso if you dont provide the software yourself, it is the duty of the devs to open it up somebody else can do. I dont care if it is a C64 or M65 Program at all. M65 is preferred though. However talking about "problems" without "solutions" usually doesnt help. Same with all other comments.

I am no dev. I just think that my requests could make the system better. If I always here - not possible - problem, ... why do I even should make those requests? If everything isnt possible or bad or whatever, I better sell it and use something else again.

You have a great foundation. Yes, a little troubled one, but I was glad to back it. I feel, that my feedback even isnt wanted actually.

lgblgblgb, 2022-05-29:
The problem that you don't understand the purpose of this issue-tracking here that it's mega65-core not software written for mega65-core, also not the c64-core (and as I wrote somewhere else c64-core is NOT the "go64 thing", go64 is still the mega65-core, c64-core is totally different separated core the so called "alternative cores" just like the uno spectrum core, or the gameboy core, but not the mega65-core we talk _here_). As it seems you don't even understand the difference between a core and a software (ROM is a software, core is a hardware, in the sense that the core creates the hardware - inside the FPGA). This is not a problem for sure, that you don't understand, everybody including me needed time to understand and not "born with the knowledge".

But to be specific and stick with the ethernet and CCGMS topic: my problem here that it's mega65-core's repository, here we talk about the problems/wishes on the mega65-core itself, not the software should be written for utilize that. Your idea is indeed very useful, as I've written, just cannot be handled here too much, developers here are focusing on the core itself, that is the sole purpose of this repository. But again, indeed, there should be a place for ideas like this, you're right, and of course I understand your intent that you only tries to make the "whole ecosystem" better with some ideas.

So I take significant amount of free time of mine to try to help you to understand, hoping that I can help you to better understand things including why it's not a good place here at least, to ask for these. As you could read, I also promised that I try to discuss it within the team that there should be a place for that. So after all of these I really cannot understand too much what's your problem with my comments, where I always want to support you in any possible way I can, including trying to help you to understand things like the difference between core and ROM, and things like that.

In any way, if my not so perfect English is a language barrier here which made me misunderstand anything you wrote, then of course I'm really sorry about that.

lydon42, 2022-05-29:
This seems to request the development of a application software. The core provides Ethernet functionality, so this can already be developed. So this is no core feature request.

closed 2022-05-29 by lydon42