MEGA65/mega65-core #567: Core file selection menu has duplicate incorrect entries

dansanderson, 2022-06-06:
**Test Environment (required)**
 - Platform: MEGA65
 - Core Commit: bdeeb15
 - ROM Release: 920364

**Describe the bug**
When installing a new core from the core menu, the core file selection screen duplicates every .cor file with a similar entry with an odd character at the beginning of the name. The "file" with the odd character is not a real core. Selecting it will warn that the device ID doesn't match, but the user can continue to write the errant file to a core slot.

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare a microSD card in the MEGA65. Move it to a PC, copy some valid .cor files onto it, then move it back to the MEGA65.
2. Turn off the MEGA65, hold NO SCROLL, then turn on the MEGA65. The core selection menu opens.
3. Press Ctrl + 2 (or any numbered slot). The core file selection screen will display.

**Expected behavior**
Only the actual .cor files should appear in the menu.

**Screenshots**
[image]

**Additional context**
I manipulate my SD card with a macOS computer. I don't think it's a Mac-specific artifact but I could be wrong.

The errant file entry does not appear when the SD card is mounted to a PC (or at least not on my Mac).

lgblgblgb, 2022-06-06:
Just a comment and guessing here: can it be an LFN issue? Ie, if not DOS short name compatible names are used (8 + 3 characters, upper case only, US-ASCII only) then an LFN entry is created for the filename as well.

gurcei, 2022-06-06:
I'm wondering if it is a file with a dot at the start of the file (hence hidden from view, perhaps visible by doing an `ls -lha`?)

dansanderson, 2022-06-06:
Aha, @gurcei wins the prize! These are macOS resource fork files with names like `._C64MEGA65-1-R3.cor`.

This is now a feature request to have the core file selector ignore dotfiles. (LFN support is always good too but for this issue it's about the dotfiles.)

lydon42, 2022-06-07:
This requires LFN support in both hyppo and qspiflash, because DOT is no valid DOS filename character.
Postponed until LFN works @gurcei

lydon42, 2023-06-09:
683-cartflash already should not show those files anymore.

lydon42, 2023-06-16:
Check with a image provided by @dansanderson shows that the changes seem to be sufficient.

closed 2023-06-16 by lydon42