MEGA65/mega65-core #569: Chase up why removing D02F from hypervisor impacted un-freezing

gurcei, 2022-06-07:
**Test Environment (required)**
 - Platform: DevKit
 - Core Commit: 7c5281e78
 - ROM Release: 920362

**Describe the bug**
A recent commit @lgblgblgb made relating to removal of D02F usage within hypervisor seemed innocent enough, but to our surprise, impacted the un-freezing process in some way that is worth investigating in future when we have a chance:
- https://github.com/MEGA65/mega65-core/commit/7c5281e782508073d9fb99c5e50a0096a63b4b4a

For now, I've partially reverted the change inside only `freeze.asm` (as that was the one impacting the un-freezing), and now un-freezing behaves normally once more:
- https://github.com/MEGA65/mega65-core/commit/899cf4f9610d86318ce7f77243e5103eb9f3c5f4

**To Reproduce**
Steps to reproduce the behavior:
1. Enter freeze menu by holding down RESTORE key
2. Then immediately exit the freeze menu with F3
3. On some occasions, I saw BASIC return, but stall (stuck in an endless loop waiting for sd-card to be ready). On another location, BASIC returned successfully, but doing a `DIR` got me `drive not ready` messages.

**Additional context**
A lot of these investigations came about when debugging alongside Dr Vulcan and assessing his issue, in this Discord thread:
- https://discord.com/channels/719326990221574164/791383472853614593/982994187681161278

lgblgblgb, 2022-06-07:
Start of the "technical" discussion in PR #568 : https://github.com/MEGA65/mega65-core/pull/568#discussion_r890547705

lgblgblgb, 2022-06-07:
By the way, it's worth a check to try this, after someone can trigger unfreeze problem: so after having problems with `DIR` etc, trying to `POKE` $D02F with the VIC-IV sequence if that makes `DIR` work then.