MEGA65/mega65-core #571: VIC-II-style multicolor char mode - wrong colors used for char color >= 8

drwuro, 2022-06-07:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65 vs. xemu
 - Core Commit: 7c5281e
 - ROM Release: 920365

**Describe the bug**
When selecting one of the upper 8 colors as the "char color" in multicolor mode, the color displayed should be one of the 8 primary (lower) colors (black to yellow), with the highest bit enabling multicolor mode for this char. However, in the current MEGA65 core, multicolor mode is enabled, but still the "upper" color is displayed.

It does however work as expected in C64 mode.

I asked Paul whether this behavior is intended (to make sure there's not another flag/register that I've overlooked that enables this functionality by default in the VIC-IV - in order to get more colors), and he confirmed on Discord that this is faulty behavior.

**To Reproduce**
Steps to reproduce the behavior:
1. Enter POKE $D016,$D9 to enable multicolor mode
2. Press MEGA + 4 to select medium grey color, which should be CYAN in multicolor mode
3. Type "SHIFT + * (asterisk)" a few times
-> on the MEGA65, the line is GREY instead of CYAN. in xemu, the line is cyan (expected C64 behavior)

**Expected behavior**
see "Describe the bug"

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
-

hernandp, 2022-06-07:
I can look into this

johnwayner, 2022-06-08:
@hernandp @drwuro 
The viciii_extended_attributes (bit 5 of $D031) is causing the foreground color to be 8 bit instead of 3.  If you unset $D031.5, then it works as expected.  I'm not sure if @gardners forgot about that attribute bit, or if the VHDL is incorrect.  Here's the relevant bit:
```
 if viciii_extended_attributes='1' then
                      -- multi-colour text mode with full-colour VIC III/IV mode
                      paint_foreground <= glyph_colour;
                    else
                      -- multi-colour text mode masks bit 3 of the foreground
                      -- colour to select whether the character is multi-colour or
                      -- not.
                      paint_foreground <= "00000"&glyph_colour(2 downto 0);
```

johnwayner, 2022-06-08:
Ah it was changed due to this issue: https://github.com/MEGA65/mega65-core/issues/420

johnwayner, 2022-06-08:
@drwuro Sorry, but I don't fully understand your comment.  Regardless, I don't have the history to really answer anyway.  I can however provide this documentation from The Mega65 Book that seems to agree with the VHDL:

_The C65 / VIC-III attributes and the use of 256 colour 8-bit values for various VIC-II colour registers is enabled by setting bit 5 of $D031 (53297 decimal). Therefore this is highly recommended when using the VIC-IV mode, as otherwise certain functions will not behave as expected._

gurcei, 2022-06-08:
I'm wondering then, with @johnwayner's findings, should this $D031.5 bit be cleared by the ROM upon the user entering C64 mode with GO64?

drwuro, 2022-06-08:
yes it works fine in C64 mode. however the flag seems to be set by default in MEGA65 mode - which MIGHT be fine and intended.

drwuro, 2022-06-08:
...ah, but now, the problem is that you cannot use one of the primary (<8) colors in multicolor mode (because if you do, the char will be displayed in hires mode, just like on the C64)...

it's tricky

johnwayner, 2022-06-11:
If viciii_extended_attribute mode is enabled ($D031 bit 5 = 1) AND vicii multi color mode is enabled ($D016 bit 4 = 1), then, based on issue #420, I think that the intended behavior is that 8 bits of color ram should be used to index into the system palette when the screen ram half nibble is '11'.

The issue is that after the logic I posted above that uses the full 8 bits for the color index, is another bit of logic that determines which mode the drawing will happen in:
```
                if text_mode='1' and glyph_colour(3)='0' then
                  -- Multi-colour text mode only applies to colours 8-15,
                  -- so draw this glyph in mono
                  paint_fsm_state <= PaintMono;
                else
                  -- Really do low-res multi-colour
                  paint_fsm_state <= PaintMultiColour;
                end if;
``` 

I think this should be:

```
                if text_mode='1' and viciii_extended_attributes='0' and glyph_colour(3)='0' then
                  -- Multi-colour text mode only applies to colours 8-15,
                  -- so draw this glyph in mono
                  paint_fsm_state <= PaintMono;
                else
                  -- Really do low-res multi-colour
                  paint_fsm_state <= PaintMultiColour;
                end if;
```

Thus, if vicii_extended_attributes is enabled, the PaintMultiColour draw state will be used and the entire 8 bits will index into the color palette.  Without this change, if the 8 bit color index happened to have bit 3 = 0, it would force a PaintMono.

I've built a bitstream with this change and it seems ok to me.

@gardners or @TheChief (since you submitted the fix for #420) If you agree with this assessment I can submit a PR.

TheChief, 2022-06-11:
I'm fine with it, let's give it a try.

johnwayner, 2022-06-14:
Well, I've sat on this for a few days.  I made the change locally and it is weird that you can no longer do hi-res mode when the extended attributes is active. There are some choices here for how to handle when multi-colour and extended attributes are both enabled:

1. Always use the full colour ram byte as an index into the palette giving 256 possible colours at the expense of having every character forced into multi-colour mode (two bytes per wide pixel).
2. Use the 7th bit of the colour ram to determine whether the character is in multi-colour mode or high-res (just like bit 3 of colour ram in VIC 2 multi-colour mode).  This maintains the ability to mix high-res and multi-colour graphics, but limits the colours to only those between 0-127.
3. Totally ignore extended attributes if multi-colour is enabled.  This seems to go directly against the fix @TheChief made for #420.  Bit 3 chooses hi-res/vs multi-colour, and the color values would be limited to 0-7.

I feel a little strange deciding this here.  It seems like this is something the C65 already had to decide, but I haven't been able to figure out what the VIC 3 does in this case.  I suspect it does option 3 since the c65 manual at zimmers doesn't seem to mention any changes to VIC 2's multi-colour mode.

Option 2 seems the most flexible and reasonable extension of VIC 2's MCM.  But I'm so new to this stuff that I don't feel comfortable making this call.

gardners, 2022-06-17:
So, what I think makes the most sense here, is that when VIC-III extended attribute mode is enabled, bit 3 still selects multi-colour mode, but the upper bits are still available for colour. i.e., there are 128 multi-colour and 128 hi-res colours, just that bit 3 rather than bit 7 is the selector between them.  That is, you can still do everything the VIC-II let you do, but you have more colours available to you in the process.

gardners, 2022-06-17:
The current behaviour is thus, correct. By using bit 3 in the colour lookup, it is _possible_ to select different colours for the MCM and mono chars, but it is also possible to structure your palette so that they are the same for both, as with the VIC-II.  

At most, I would mask bit 3 out of the colour look up, if MCM mode is enabled, when VIC-III extended attributes are enabled, if people think that that behaviour would be (a) less surprising (which I think is true); and (b) result in no loss of functionality that folks might want (i.e. different FG colours for MCM vs mono chars).

gardners, 2022-06-18:
Have opted for the less-surprising behaviour of allowing all colour bits in VIC-III/IV mode, except bit 3, which gets masked out, so that it behaves in a manner that C64 programmers will expect.

gurcei, 2023-03-20:
Ok, I feel like gardners' fix was sufficient to address this card and we can close this here.

TheChief has offered an alternate solution in his PR #657.

So discussions on alternate/better approaches can continue there.

gurcei, 2023-03-20:
Acutally, nah, as his PR references this issue, let's keep this issue open until his PR is sorted out :)

gurcei, 2023-04-23:
I've merged in @TheChief 's PR into development on this, so will close.

closed 2023-04-23 by gurcei