MEGA65/mega65-core #577: NO SCROLL Locks up m65

dropnine, 2022-06-12:
**M65 locks up when listingt BASIC in 1MHz speed.** 

Plastform: DevKit, MEGA65
firmware: ce23c67
ROM: 920366

To reproduce:
1. SPEED 1
2. LIST
3. Press NO SCROLL
4. Press RUN/STOP

Additional Info
You can recover with RUN/STOP - RESTORE. 
Does not seem to affect 3.5 or 40 speeds.
Affects botrh 40 and 80 column mode.
Locks up on any line that no scroll is pressed.

lydon42, 2022-10-02:
Can't confirm this with the Batch 2 RC1. NO-SCROLL does stop the listing, pressing No-SCROLL again continues listing just fine.

dropnine, 2022-10-02:
The issue still exists in the newst releases.  

After pressing NO SCROLL, pressing RUN/STOP breaks from the listing in all speeds but in 1MHZ mode; it locks up the M65.

Instead of pressing NO SCROLL a second time, **press RUN/STOP** as Step 4 points out.

lydon42, 2022-10-02:
Ok, makes sense now, thanks for the clarification. So now we need to determine if it is a core bug or if it has it's roots in the rom.

dropnine, 2022-10-04:
For sure lol

If there's anything I can do to help, please let me know.

Cheers.

lydon42, 2022-10-04:
Just to avoid later confusion: this is probably a ROM issue, as NO SCROLL has no special core functionality.
We keep it open here until confirmation.

dansanderson, 2022-11-07:
https://github.com/MEGA65/mega65-rom-public/issues/25#issuecomment-1305238506

dansanderson, 2023-08-06:
This has been fixed in the ROM. This issue can be closed. (I don't have contributor access to this repo so I can't close it.)

dropnine, 2023-08-06:
Tested issue.  Seems to be solved.  Speed 1 lock up in respect to no scroll, Run/stop works. Pressing caps/lock unlocked mega65 tested and is also resolved and working fine.

closed 2023-08-06 by dropnine