MEGA65/mega65-core #579: Tiled sprite not covering right-most pixel of the screen

zendar, 2022-06-12:
- Platform: NexysA7
 - Core Commit: d71fa6b
 - ROM Release: 920352

**Describe the bug**
When using 200px high, tiled sprite, the rightmost pixel-column is visible indepent of what Xpos of sprite is. The column is so thin it looks like a 1pixel column of 640px wide mode, even though I am running program in 320x200 mode.

Using sprite setup:

    // Enable a full color, tiled sprite with 200 height
    lda #$01
    sta $d015     // Turn on sprite
    sta $d06b     // Full color
    sta $d055     // Variable height
    lda #$f0
    sta $d04d     // Horisontal tiling (SPRTILEN)
    lda #200
    sta $d056     // Height of sprite (SPRHGHT)

    lda sprite_wipe_x
    sta $d000
    lda #50
    sta $d001
    //MSB
    lda sprite_wipe_x_msb
    sta $d010

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Tiled sprite is expected to cover whole screen from X-pos to right side of screen.

.

[image]

gardners, 2022-06-13:
Probable cause is this section of sprite.vhdl:

```
    -- Stop drawing sprites in right fly-back, to prevent glitches with
      -- horizontally tiled sprites.
      -- (but allow non-tiled sprites to wrap around into left border, as on the
      -- C64.)
      if (sprite_h640='0') and (x_in > 416) and (sprite_horizontal_tile_enable='1') then
        x_in_sprite <= '0';
      end if;
      if (sprite_h640='1') and (x_in > 832) and (sprite_horizontal_tile_enable='1') then
        x_in_sprite <= '0';
      end if;
```

Can someone please confirm if the problem exists in both 40 column and 80 column mode?
Also, if someone can please produce a minimum failing test in mega65-tools/src/tests/ that uses the VIC-IV pixel-read-back to confirm if it is happening or not in an automatic way, this will help regression testing.

gardners, 2022-06-13:
Problem is _not_ that the sprites stop one pixel early, but something else stranger, as the VIC-IV pixel colour read-back thing can't read the values. Also note that it is only on every second raster.  I'm suspecting that the sprites might stop rendering 2/3 of a pixel early at this stage. Currently investigating.

commit 7a3a7025 (2022-06-14): fix #579
closed 2022-10-09 by lydon42