MEGA65/mega65-core #581: VIC-IV border registers show no effects on Nexys/MEGA65 - work fine with xmega65 emulator

snoopy-f64, 2022-06-15:
**Test Environment (required)**
 - Platform: Xemu vs. Nexys/MEGA65
 - Core Commit: 191
 - ROM Release: 920366

**Describe the bug**
Writing to VIC-IV registers $d048/49 (TBDRPOS), $d04a/4b (BBDRPOS), $d05c/5d (SDBDRWD) (line 140 and 150 of attached prg) has no border (moving) effects using Nexys or MEGA65.

Works fine using emulator (xmega65).

**To Reproduce**
[bordertest.prg.zip](https://github.com/MEGA65/mega65-core/files/8907151/bordertest.prg.zip)

**Screenshots**
[image]

[image]

lydon42, 2022-10-02:
I see no problem on RC1. Please recheck.

hernandp, 2022-10-11:
This must be some kind of regression as I tested against hardware  when I
implemented the border registers, and they worked OK.