MEGA65/mega65-core #584: Complete paddle support

dansanderson, 2022-06-18:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 380f41a
 - ROM Release: 920365

**Describe the bug**
Someone in the Discord mentioned that paddle support is incomplete. I did a quick test and am opening this issue to track completion of this feature. (I don't know if this is core or ROM.) Issues I noticed:

1. Paddles are not recognized on joystick port 2.
2. JOY(1) and JOY(3) both return the same paddle on joystick port 1, and JOY(2) and JOY(4) return the other one.
3. There is about a quarter turn at the end of the counterclockwise direction that all reads as 255, so the entire range of motion is not reported across 0-255. (I don't know if this might be an issue with my paddles or if this is expected for Commodores.)

**To Reproduce**
Steps to reproduce the behavior:
1. Use this BASIC program: `10 PRINT JOY(1), JOY(2), JOY(3), JOY(4):GOTO 10`
2. Connect Atari-style paddles to joystick port 1. Run the program, and adjust both paddles all the way clockwise and counterclockwise.

* JOY() reports 0 when all the way clockwise, and increases steadily to 255 as turned counterclockwise.
* JOY(1) and JOY(3) report the same paddle on port 1, and JOY(2) and JOY(4) report the other paddle on port 1.
* On my paddles, there is about 90 degrees of slop between reading 255 and the actual full counter-clockwise position (it reads 255 the last quarter turn). I don't know if this is a hardware issue with my paddles or something worth fixing in the core/ROM, and I don't know how this corresponds to vintage hardware.

3. Connect Atari-style paddles to joystick port 2. Run the program, and adjust both paddles all the way clockwise and counterclockwise.

* No effect. All four JOY() functions return 255.

I'm happy to test candidate fixes, just let me know.

dansanderson, 2022-06-18:
(ROM code looks ready on visual inspection, so I'm guessing this is a CIA paddle port wiring issue in the core.)

lydon42, 2022-10-02:
As paddles do work with the C64 Core, it certainly should be fixable.

dansanderson, 2023-01-27:
I'm going to nuke this and convert it to a ROM issue. I'm not aware that more wiring is needed to support paddles connected to port 2. The POT() function is not working to spec but the cause is not clear. (I don't know why I said "JOY" above, I clearly meant "POT". But no matter. :) )

Also, I only recently just learned that the "missing" quarter turn is because I'm using Atari paddles. Commodore paddles use a different potentiometer value. I acquired some Commodore paddles today and they correctly register 0-255 across the entire turn.

closed 2023-01-27 by dansanderson