MEGA65/mega65-core #585: Some bitstream builds result in faulty attic ram reads

johnwayner, 2022-06-19:
**Test Environment (required)**
 - Platform: MEGA65
 - Core Commit: 380f41a  (aka build v192)
 - ROM Release: Any

**Describe the bug**
Inconsistent values read from attic ram ($8000000).

**To Reproduce**
Steps to reproduce the behavior:
1. Type 'MONITOR' <return>
2. Type 'M 8000000' <return>
3. Scroll back to the 'M 8000000' and hit return again
4. Do step 3 several times. 

The values shown in some locations will change with each viewing.

**Additional context**
This seems to be an issue synthesizing the bitstream.  A rebuild of the same VHDL for the next build (v193) provided a core without this issue.

I'm creating this Github issue mainly to get an issue number for a mega65-tools automated test.

lydon42, 2023-10-29:
test_585 still fails at the second test address in the range, as it did before.

lydon42, 2023-11-09:
Merged 585-hyperram-bugs into development, this fixes the tests in test_585 and test_710 on r3 and r4 boards.

closed 2023-11-09 by lydon42