MEGA65/mega65-core #589: Core menu can be entered by powering up with space-bar held

gurcei, 2022-07-01:
My daughter discovered this by her own accidental bash-testing.

I even recall on one of her bash-testing forays, she managed to jump straight into the flashing mechanism (skipping the initial core slot menu altogether).

Part of me feels like we might be able to replicate these issues with some bash testing ourselves.

gurcei, 2022-07-01:
Other observations:
- While on the core slot menu, if I press c, b, d it surprisingly goes to a file chooser (to pick a core).

- pressing 'm' shows the flash monitor, but @gardners explained this is intended behaviour

lydon42, 2022-08-17:
Also TAB and Joystick buttons seem to be (or have been) an entry vector. This needs to be cleaned up further. Perhaps also differentiated by target platform, so we reduce the keys for the *real* thing, but leave in the backdoors for the development boards, where you might have sub-optimal USB keyboards attached.

lydon42, 2022-08-17:
Also: add display of flasher/core version to the flasher screen.

closed 2022-09-24 by lydon42