MEGA65/mega65-core #591: External RTC advances time while on, but time is still read from faulty internal RTC on boot

dansanderson, 2022-07-08:
**Test Environment (required)**
 - Platform: MEGA65
 - Core Commit: 80fc1c0
 - ROM Release: 920371

**Describe the bug**
I have a batch #1 MEGA65 with a faulty internal Real-Time Clock (RTC) that does not tick. I also have the RTC battery installed. I acquired an external RTC and Grove cable, and now when the MEGA65 is on, time advances successfully. However, when I turn off the machine and turn it on again, the system clock resets to the most recent time I set in the Configuration menu. I suspect it is reading the time from the faulty internal RTC.

**To Reproduce**
With a MEGA65 that has a non-ticking internal RTC (as reported my MEGAINFO):

1. Install the external RTC with Grove cable.
2. Turn on the MEGA65. Ensure that it is running core 80fc1c0 (v201) or later; upgrade core slot 1 to such a core if it isn't. Confirm with MEGAINFO (with recent SD files, Freeze menu then Help key) or another method that the clock is ticking successfully.
3. Turn off the MEGA65. Hold Alt, turn on, then enter Configuration (1). Set the time and date, then save and exit.
4. Turn off the MEGA65, then turn it on again to confirm it is running the latest core from slot 1.
5. Wait a few minutes. Check the system time (such as with `?TI$`) to confirm it is advancing.
6. Turn off the MEGA65, then turn it on again. Notice the time reported in the BASIC title screen, or `?TI$` again. Notice that it has reverted to the previously-configured time, and did not preserve or continue advancing the time while turned off.

**Expected behavior**
The actual system time should be preserved and advanced by the battery-powered RTC.

grim-fandango, 2022-07-08:
Even though the hours are stuck at 19 on the MEGA65, I've noticed that the date advances still - which suggests that the RTC has a sense of what the hours are.  My guess is that the MEGA65 always sets the hours to 19 and reads the hours as 19.  I'll test this and report back

dansanderson, 2022-07-08:
I have confirmed that the battery of my DS3231 is only reading about 0.8V. I will pursue the recommendation to use a soldering iron to connect a more reliable 3V source, then re-test.

dansanderson, 2022-07-09:
I have installed core 198 (f555316), and have upgraded the battery on my DS3231 to a CR2032 that measures 2.6V with the same polarity as the previous battery. I have used MEGAINFO to confirm that it is using the external RTC and it is ticking. I have used the Configuration utility to set the time (17:32:40 as I write this); doing so boots into the factory core, so I power off, then power on. The time now reads 19:35:11.

I am now in full agreement with @grim-fandango : There seems to be something peculiar setting the hour to 19, while preserving the other date and time values. Everything else seems to be working with the external RTC, including battery backup of the time. This is true even with the battery on the main board removed.

dansanderson, 2022-07-09:
(Updated issue title and description.)

dansanderson, 2022-07-09:
I left the MEGA65 powered on starting at "19":37:34, and the time now reads "19:04:18". So the "19" is being forced in real time; it doesn't tick over to 20.

gardners, 2022-07-09:
Please try POKE $FFD7402,<something other than $19> from BASIC65, and report if it changes the hours.

dansanderson, 2022-07-09:
```
POKE $FFD7402,$03
?TI$
03:40:59
```

Yes, the POKE updates the hours. This value survives the reset button, but not a power cycle: turning the machine off then on again sets the hour back to 19.

dansanderson, 2022-07-09:
New data point: I set the hour to $03 with the POKE and let the minutes roll over. The hour value *stayed* at 03.

```
?TI$
03:49:24

?TI$
03:01:38
```

grim-fandango, 2022-07-09:
Well, I watched the date all day yesterday, and 15 hours later it still hadn't changed, but then I was just too tired to wait another hour :-)  This morning the date has changed, so at some point the date has incremented.  I've just tried setting the time to 2355 to see if the date flipped at 'midnight', but it didn't.  Very peculiar!

grim-fandango, 2022-07-09:
Another observation: when in MegaInfo, it detects the RTC, and says "CHECKING".  It then starts displaying the time.  If you want until the top of the hour, it says CHECKING again, when you wouldn't expect it to.  Just noting that here in case that was an indicator of something

dansanderson, 2022-07-09:
A clue from a troubleshooting session with Paul in Discord: the hour value survives the reset button, but not the FPGA reset poke. This indicates bad initialization.

Set the hour value and confirm it is set:
```
POKE $FFD7402,$10:?TI$
```

FPGA reset into core slot 1:
```
POKE $FFD36C9,$40:POKE $FFD36CF,$42
```

This resets the hour to 19.

We could see this from the Alt menu via the serial monitor: set the hour value again, power off, hold Alt, power on to get to the Hypervisor menu with core 0. Hold Alt on the MEGA, then In the serial monitor:

```
sffd36c9 40
sffd36cf 42
```

This resets to the menu again with core 1. Confirm this in the serial output, then test the hour value with `mffd7402`. I saw:

```
MEGA65 Serial Monitor
build GIT: development,20220623.13,f555316

.:0FFD7402:1901080722000000000000001C08001C
```

(See the "19" at the beginning.)

dansanderson, 2022-09-14:
FYI this issue persists in RC 0.95 core dadad1c. (I'm not aware that anyone submitted a fix, just saying it will continue to be an issue if not fixed for RC 0.95.)

lydon42, 2022-09-25:
> Another observation: when in MegaInfo, it detects the RTC, and says "CHECKING". It then starts displaying the time. If you want until the top of the hour, it says CHECKING again, when you wouldn't expect it to. Just noting that here in case that was an indicator of something

This is expected, MEGAINFO can only do between 1-2 hours of tracking, then the counter will run over and it will start detecting again.

pietpukkel, 2022-09-25:
it however also does this when starting MEGAINFO just 1 minute before rollover. Reverts to checking and the hour does not rollover.

lydon42, 2022-09-25:
F1 enables debug output in the bottom row...

lydon42, 2022-09-25:
And to summarize my investigation today: I can't see any place where the RTC hours get set to 19, nor where the RTC gets totally reset. Looked into hyppo and i2c vhdl.

lydon42, 2022-09-27:
The part where the 19 can be find is the signal initialization in grove_i2c.vhdl:
```
  signal write_job_pending : std_logic := '0';
  signal write_reg : unsigned(7 downto 0) := x"02";
  signal write_val : unsigned(7 downto 0) := x"99";
```

But write_job_pending is 0, so it is unclear to me why it is writing it on startup.

Quick fix will be implemented by changing write_reg to 0d, which will write to the second alarm day register instead. As alarm features are unused, this will solve the clock hour 19 problem, but it is obviously not the final solution to the problem.

closed 2022-10-03 by lydon42