MEGA65/mega65-core #594: Sprite fetch  effective address  incorrect

hernandp, 2022-07-24:
**Test Environment (required)**
[image]

**Describe the bug**
(Run in 64 mode)
* Setup MEGA65 I/O Mode.
* Set screen pointer to 0x008C00
* Set sprite table pointer to 0x008C00 + 0x3F8
* Set sprite 0,1,2 address to 32   (32  * 64 = 0x800)

According to documentation:
* If default bank is 3 (base address $0000) and VICII banking is considered in sprite fetch effective address,  Sprite 0 will be read from $800.
* If VICII banking is ignored, and SPRPTR16 is not used, I expected address fetch to be also from $800

But copying sprite data to $8800 will display the sprite 0 properly.  This is I think, Incorrect (or undocumented) behavior.

**To Reproduce**

Use the attached C program to play, or use the provided BANK65.PRG.

[bank65.zip](https://github.com/MEGA65/mega65-core/files/9175069/bank65.zip)

**Expected behavior**
Sprite 0 must be displayed as garbage from $800.

gardners, 2022-07-25:
The sprite data address always comes from the same VIC-II bank as the sprite pointer address, according to the VHDL (see around line 4980 of viciv.vhdl). What does the documentation say?

hernandp, 2022-07-27:
Documentation just says: ". This is done by setting the SPRPTR16
signal (bit 7, $D06E, decimal 53358), which tells the VIC-IV to expect two bytes per
sprite pointer instead of one. These addresses are then absolute addresses, and ignore
the 16KB VIC-II bank selection logic. "

That's quite understandable: SPRPTR16 will make VICII bank to not affect sprite fetch address.

But how address generation works when SPRPTR16 is OFF,  the sprite pointer register selects e.g: to put the table at $50000 and VIC-II banking is active.   What I found counterintuitive is why the sprite pointer table should affect the sprite pointers themselves, as the documentation states the $D06C - $D06E  is just for selecting the table location in memory, not the addresses of the sprites.

In other words, if I have the default pointer table, with default bank ($DD00=3) at screen+$3F8  with the first entry set at 13 that yields address 832 for sprite #0.  Thats correct I guess.

But if I set $D06E=$05, $D06D=$00, $D06E=$00 to set the pointer table at $50000, first entry to value 13,  and I'm still with default VICII bank  ($DD00=3),   why it wont fetch also from 832 ?  That's not stated in documentation.

I hope that I'm clear with my doubts.  
Regards,
hernan

hernandp, 2022-08-30:
@gardners  I understood.   Bits 14 / 15 can be set by the Sprite pointer VICIV registers $D06C..0E  or by $DD00 if HOTREGS are enabled. My confusion was that $DD00 was the unique responsible -always- to set banking bits (14-15).  I think that is can be closed.   We could review documentation to see if this is stated clear.  
XEMU fix coming along.
 Thank you!.

closed 2022-10-02 by lydon42